cortex-lab/KiloSort #107: Assertion error

ingridvanwelie, 2017-12-11:
Hi, I just started using kilosort and phy and generally, kilo sort runs well on my data. However, I am running into issues with the template gui. When I try and open it, I get the following assertion error:

(phy) Ingrids-MacBook-Pro:Ingrid lawelie$ phy template-gui params.py
10:09:21 [E] cli:39               An error has occurred (AssertionError):

Any suggestions as to what the issue may be? As far as I can tell, spikes were found and I didn't change anything in the config file since running a previous file through kilosort and reviewing it in phy without problems.

nsteinme, 2017-12-11:
Try using "phy template-gui params.py --debug" to get more information
about the problem?

ingridvanwelie, 2017-12-11:
Thanks! It seems to have a problem with the channel mapping? 

File "/Users/lawelie/miniconda3/envs/phy/lib/python3.5/site-packages/phycontrib/template/gui.py", line 331, in _init_data
    assert np.all(channel_mapping <= self.n_channels_dat - 1)
AssertionError

nsteinme, 2017-12-11:
I think self.n_channels_dat comes from your params.py file, and
channel_mapping will come from channel_map.npy. It seems to be saying that
your channel_map.npy has a number in it that's larger than the number of
channels in the data file, as specified by params.py. So check those two
files and see whether they are accurate, or whether they provide
conflicting information?

ingridvanwelie, 2017-12-11:
Yes, I think I found the mistake - the channel number in the params file was wrong. 

Thanks for the quick feedback! Looking forward to starting sorting ;-).

closed 2017-12-11 by nsteinme