cortex-lab/KiloSort #109: Crash during fitTemplates -- inner matrix dimensions

weallen, 2017-12-30:
Hi,

On one dataset, KiloSort is for some reason crashing part way through with the following error:
connected to 24 workers.
Time 402s. Optimizing templates ...
Time 425.12, batch 381/19518, mu 10.00, neg-err 62156.758079, NTOT 1068794, n100 2752, nTime 508.42, batch 601/19518, mu 15.15, neg-err 148770.892217, NTOT 1329287, n100 3431, n200 2581, n300 1993, n400 1228
Error using  * 
Inner matrix dimensions must agree.

Error in fitTemplates (line 192)
    data = dataRAW * U(:,:);

Error in sortNpix (line 19)
rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively

The matrices on crash are of sizes:
K>> size(U)

ans =

   372   960     3

K>> size(dataRAW)

ans =

       32832         195

anupamgarg, 2018-03-01:
Any update on what might be causing this issue? Also having the same issue when running Kilosort on a large dataset (~500GB).

nsteinme, 2018-03-02:
For the specific example Will has given, it seems that the data came out as
having 195 channels but the templates expect to cover 372 channels. So
check the channel map, check ops.NChan and ops.NChanTOT and make sure these
are all correct.

Given that you have a really large datafile, another possibility is that
the hard drive ran out of space when writing the temporary data file that
it uses. You can set the location for temp.wh to be on any other hard
drive, and you may need to do that in this case.