cortex-lab/KiloSort #110: From .plx to .dat to kilosort

eduardoartea, 2018-01-09:
Hi,

I am recording neural data using the plexon system, so it is adquired as a .plx file. There are SDK from plexon to take that data and load it into Matlab. I am taking the variables were the voltage from every electrode is recorded to make a .dat file. I use that file and run kilosort. However, when the analysis is done, nothing looks like an spike in the image that kilosort shows when is finished, nor in Phy.

I wonder if I am not saving correctly making the .dat file, or if there is something wrong in the way I am using kilosort which is: 

1.- I made a ChanMap.m.
2.- Set the paths to StandardConfig and master_file.
3.- Run master_file

The questions are: Is there a easier way to go from .plx files to kilosort? Am I using kilosort all wrong? If yes, how I should do this? 

Cheers,
Eduardo

nsteinme, 2018-01-13:
Hi Eduardo,

Sounds like you are approaching the problem in the correct way. The most
likely problem is that you have constructed the dat file incorrectly, or
specified parameters of it incorrectly. It needs to be int16 format, and
size [nChan nSamples], and can be created by writing the data directly with
fwrite. Make sure that ops.NchanTOT in kilosort, and "n_channels_dat" in
phy's params.py, both have the correct number of channels specified,
otherwise it will be loaded incorrectly. To test whether kilosort has it
correct, put a breakpoint early in the processing after some data has been
loaded into the "data" matrix and plot some rows of that matrix to see if
they really do look like your data. In phy, the way to test it is to look
at traceview and see whether the traces look like your data.

hth -

xf123chen, 2018-01-29:
suggest"Matlab Offline Files SDK"

fspinar, 2018-06-11:
Hi Eduardo.
 It looks like I have the same problem.( My gui is not getting the result from matlab.) Could you solve it? If so, may I ask you, how did you fix it ?
 Thank you.

xf123chen, 2018-06-12:
font{
        line-height: 1.6;
    }
    ul,ol{
        padding-left: 20px;
        list-style-position: inside;
    }

    

plx data should be converted this format(*.dat)."Matlab Offline Files SDK"(from plexon website) should be used.

    
        a#ntes-pcmail-signature-default:hover {
            text-decoration: underline;
            color: #199cff;
            cursor: pointer;
        }
        a#ntes-pcmail-signature-default:active {
            text-decoration: underline;
            color: #246fce;
            cursor: pointer;
        }
    

                
                    
    
        font{
            line-height: 1.6;
        }
    

    
        font{
            line-height: 1.6;
        }
    

    
        font{
            line-height: 1.5;
        }
    
-----------------------------------------chen xfLaboratory of Learning and Memory, Key Laboratory of Animal Models and Human Disease Mechanisms, Kunming Institute of Zoology, CAS, Kunming 650223, China ----------------------------------------

fspinar, 2018-06-12:
Not sure what this response means.
However, just to add, we are able to read our file into Matlab and sort using KS. However, the problem is that following this, the result can not be visualized in the Phy GUI.

huangkang314, 2018-12-14:
Have a try with KilosortPLX (https://github.com/madeleinea/KilosortPLX). I successfully loaded my data, hope it will help you.