cortex-lab/KiloSort #112: Sort only part of the data

cindyhfls, 2018-01-19:


cindyhfls, 2018-01-26:
Hi Nick! @nsteinme 

I was just wondering about the input and output format of the kilosort code.

Since I used a Ripple system, my data is saved in .ns5 and .nev files, where they have analogue channels and raw channels etc. I want to figure out which part of the data I really need to extract into the binary format. Is it the microvolts data or the time or both that I needed?

Also, while the output of the Kilosort result is best visualized with a PHY-GUI. Is is possible to view it in other software, e.g. Plexon offline-sorter, neuroexplorer etc.?)

Thanks

Cindy

nsteinme, 2018-01-27:
Hi, you need to provide kilosort with the raw recorded continuous traces. It will find the spike shapes and spike times. 

It should be possible to visualize in other software, but you will need to convert the output format. See [here](http://phy-contrib.readthedocs.io/en/latest/template-gui/#installation) for a description of exactly what kilosort produces.

cindyhfls, 2018-01-29:
Thanks!
I have tried extracting raw channels from four channels and run them through the algorithm and it looked very bad. 

The raw data from the DATA matrix output from kilosort looked fine, but the temp_wh.dat (after whitening?) was empty.

I was wondering if I that is because I need to change  ops.intialize to 'fromData' or some other parameters?

nsteinme, 2018-01-30:
Hm, I'm not sure where it has gotten lost. You're right that temp_wh.dat should have whitened data. I'm not sure ops.initialize will help but you could try it. Maybe also double-check that the channel map is correct (should be just [1:4] in this case). Maybe also try ops.whiteningRange=Inf. Otherwise I think you need to work backwards from line 282 in preprocessData (where the temp.wh file is written) to figure out where the data got lost. Sorry it's hard to be more helpful...

cindyhfls, 2018-01-30:
I have the channel map correct and whitening range set to infinity. It seems that line 282 was not run because the if statement condition wasn’t satisfied. I do see CC arrays Wrot arrays although I don’t know what they meant. The uproj structure was also empty.

nsteinme, 2018-01-30:
If you can make a small dataset that reproduces the problem and post a
dropbox link to it, then I can try to take a look. I'm not sure I will be
able to do it soon though, apologies in advance...

cindyhfls, 2018-01-31:
Here's the data. https://drive.google.com/open?id=1fODOLnJVicJiRh3m7946xP-1bUds6EPT
@nsteinme

cindyhfls, 2018-02-01:
Just an update. I tried on raw data from all 32 channels and then the whitened data was still missing and I only got <50 spikes in total in the final output 

@nsteinme

cindyhfls, 2018-02-01:
[image]

cindyhfls, 2018-02-02:
@nsteinme 
Our data ranges from -40 to 200. We saw that the eMouse data range from zero to a few hundreds. Does it matter the range of data to use, or is just for simulation purposes?

Cindy

cindyhfls, 2018-02-02:
[image]

And I have been consistently observing multiple spikes in different channels even though I set the kcoords to be different for each channels...

cindyhfls, 2018-02-08:
@nsteinme Did you get a chance to look at the data?