cortex-lab/KiloSort #115: CUDA error: invalid configuration argument

stephielynn921, 2018-02-08:
The following error pops up when trying to spike sort:

Error using gpuArray/permute
An unexpected error occurred trying to launch a kernel. The CUDA error was:
invalid configuration argument

Error in fullMPMU (line 165)
            coefs           = reshape(permute(coefs, [3 1 2]), [], numel(st));

Error in master_file (line 36)
rez = fullMPMU(rez, DATA);% extract final spike times (overlapping extraction)

This error was not occurring a few days ago and was going through the files fine. I did not change any element on my end and suddenly this error occurs when running Kilosort.

marius10p, 2018-05-15:
Is this still a problem? Kilosort has no "memory", so this error must have to do with different parameters, options, maybe Matlab updates?

closed 2018-08-17 by marius10p

parviz-gh, 2020-08-27:
hi,
i have similar issue , 
Error running Kilosort! an unexpected error occurred trying to lunch a kernel. CUDA error was:
invalid configuration argument