cortex-lab/KiloSort #117: Large spikes without templates

hrapg, 2018-02-19:
We are using Kilosort to analyze our MEA data. The analysis seems to miss some of the large spikes. We attached two screen shots of the spike and the templates. In this example, the cluster was on channel 32 but we could not detect the spikes and template on channel 30. Adjusting ops.th and ops.lam did not improve the sorting. Do you have a suggestion which sorting parameter we could change to detect those large events?

[image]
[image]

nsteinme, 2018-03-02:
Do you know how many of the large spikes on channel 30 are in the whole dataset? If there are very few then kilosort may sometimes fail to find a template for them, and it should help to increase the number of templates (ops.Nfilt) you want kilosort to find. 

You may also want to see whether you can find any other templates already in the results that did find those spikes - another thing that can happen is that big spikes appear in a different cluster of small spikes when the big spikes aren't perfectly fit, and the residuals of the big spikes get detected by other templates.

hrapg, 2018-03-05:
Hi Nick,
Thank you very much for your comments and suggestions. 
We were able to find some templates of those spikes in clusters with different IDs on the other channels. For example, ID 244 on channel 27 and ID 240 on channel 34. Still only few of those large spikes were detected.
The large spikes were probably not detected by other templates as the time course of the large spikes has a significant shift in comparison to the templates on the channel 30.
Using our acquisition software, we counted 382 large spikes on channel 30. We were using ops.Nfilt equal 256 (with 60 channels in the upper range of 2-4 times, as recommended) for the sorting. After getting your comment, we repeated the sorting with a setting of 512. Interestingly, we were able to detect some large spikes on the channel 30. Still, the template was not perfectly fitting to the shape of those events and the majority of large spikes was not detected.

nsteinme, 2018-03-07:
382 spikes sounds like a small number but it depends on how long the recording was - what is that in sp/sec? If it's <0.1 or 0.01 then you may have to increase the number of templates still further, I'm not sure. 

One thing you can do it use phy to try to determine whether those spikes have been fit (but in a way that you haven't anticipated) or whether they haven't been fit at all. In traceview, with a cluster that includes some big spikes selected, use "go to next spike" to try to center the view on a big spike (or, put in the time range directly yourself for a range that includes a big spike). Then look closely at the spike: you will see the raw data as well as a second trace (dimmer) that reflects the raw data minus the sum of all the templates that were found there. If the template fits the spike well then this trace should be close to flat. If it has not fit, then you will still see the spike there. If it is flat, then maybe kilosort fit a combination of smaller templates to those spikes? I'm not sure.

hrapg, 2018-03-09:
Hi Nick,
Thank you for the suggestions. We checked the dimmer traces in traceview. The traces both showed the large spike, so we assume that these spikes were not fit.
The number of large spikes in this example is really quite low. With a recording time of 10 minutes we would still have a rate of around 0.6 spikes/sec. Could maybe the event duration have an influence on template generation? The large spikes have a quite long duration.

nsteinme, 2018-03-09:
Hm. Well, I think this is related to the low spike rate (which is hard for kilosort to find) but some other people have suggested that they think large spikes are missed sometimes. Here is the test to find out: take your dataset and make two new datasets, each with a small modification:

1) find every big spike (with a simple threshold, as you did), and multiply the values of the raw data in a small window around those spike times by 0.5. In this new dataset, the large spikes will have half the amplitude as before. Run kilosort again and see whether it detects a good (or better) template for them now. 

2) find every big spike and get the mean waveform. Then, add that mean waveform in to the trace at various random times (lots of times). In this new dataset, the large spikes will have a much higher spike rate. Run kilosort again and see whether it detects a good template for them now. 

Does this make sense? Is it a test that you think you can do? If test 1 solves the problem then it is definitely a bug. If test 2 solves the problem then it is just the low spike rate, and there's not too much to do about it. If neither of them solve the problem then we need to keep thinking. 

@marius10p There had been discussion at one point of including a feature where a "standard" spike detection is performed (either on the data, or on the residuals after subtracting all detected templates?) and any spikes that weren't otherwise detected would be put into a "others" cluster, so that at least the user would get them detected (perhaps into multiple local "others" clusters based on peak waveform location). Did you ever implement that?

marius10p, 2018-08-17:
Dropping some large spikes is a known issue, related to the initialization of Kilosort. This is fixed in Kilosort2, but it won't get fixed in Kilosort1 because it's a major algorithmic overhaul.

closed 2018-08-17 by marius10p

brendonw1, 2018-08-18:
Where is Kilosort2?

marius10p, 2018-08-18:
It will be released with the preprint, I'm still testing it.

brendonw1, 2018-08-18:
Oh great... I thought I had missed it :).  Didn't mean to be rude.  Excited
to see it all.  Is there a run-down of new anticipated features anywhere or
should we just wait for the preprint?

Thank you

Brendon