cortex-lab/KiloSort #120: Opening params.py gives 'Aborted!' error

shihaisun, 2018-03-01:
Hi all,

I got an error while looking at my KiloSorted data through Template-GUI. Every time I try to open it up again I get an "Aborted!" message. I have attached a screenshot of the output from cmd leading up to the error. Does anyone know how to resolve this?

[image]

Thanks,
Scott

nsteinme, 2018-03-02:
Hi, please take this issue to the phy-contrib github page:
https://github.com/kwikteam/phy-contrib/issues

When you do, please try running it with the --debug flag:
> phy template-gui params.py --debug

and provide that output. Thanks -

Sarruedi, 2018-05-01:
I also ran into the same issue with one of my data sets and wonder what's the cause of the error.
So far I never ran into this error. Did anybody find a workaround for this error by now?

Here's the debug log for my case:
(phy) F:\SGL_DATA\110817>phy template-gui params.py

Aborted!

(phy) F:\SGL_DATA\110817>phy template-gui params.py --debug
11:35:05 [D] config:46            Load config file `C:\Users\SRL_KiloSort\.phy\p
hy_config.py`.
11:35:06 [D] plugin:35            Register plugin `phycontrib.kwik_gui.gui.KwikG
UIPlugin`.
11:35:06 [D] plugin:35            Register plugin `phycontrib.template.gui.Templ
ateGUIPlugin`.
11:35:06 [D] plugin:35            Register plugin `phycontrib.utils.precache.Pre
cachePlugin`.
11:35:06 [D] plugin:35            Register plugin `phycontrib.utils.saveprompt.S
avePromptPlugin`.
11:35:06 [D] plugin:72            Scanning `C:\Users\SRL_KiloSort\.phy\plugins`.

11:35:06 [D] cli:107              Attach plugin `phycontrib.kwik_gui.gui.KwikGUI
Plugin` to CLI.
11:35:06 [D] cli:107              Attach plugin `phycontrib.template.gui.Templat
eGUIPlugin` to CLI.
11:35:06 [D] gui:253              Loading traces at `MSC17-68_PPC_SOM_HitMiss_Be
havior_D2500um_Rec2_g0_t0.imec.ap.bin`.
11:35:06 [D] gui:268              Loading amplitudes.
11:35:06 [D] gui:277              Loading 7675091 spike clusters.
11:35:06 [D] gui:283              Loading spike templates.
11:35:06 [D] gui:289              Loading spike samples.
11:35:06 [D] gui:293              Loading templates.
11:35:06 [D] gui:299              Loading the whitening matrix.
11:35:06 [D] gui:303              Loading unwhitened templates.
11:35:06 [D] gui:321              Loading similar templates.
11:35:06 [D] gui:326              Loading channel mapping.
11:35:06 [D] gui:334              Loading channel positions.
11:35:06 [D] gui:339              Loading features.
11:35:06 [D] gui:370              Loading template features.
11:35:06 [D] gui:424              HP filtering the data for waveforms
11:35:06 [D] gui:453              Loading the cluster groups.
11:35:06 [D] context:56           Initialize joblib cache dir at `F:\SGL_DATA\11
0817\.phy`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_mean_masks`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_mean_features`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_mean_waveforms`.
11:35:06 [D] context:80           Load memcache for `phycontrib.template.gui.get
_waveforms_amplitude`.
11:35:06 [D] context:80           Load memcache for `phycontrib.template.gui.get
_waveform_lims`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_feature_lim`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_trace_lim`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_close_clusters`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_probe_depth`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.get_correlograms`.
11:35:06 [D] context:80           Load memcache for `phy.cluster.manual.controll
er.spikes_per_cluster`.
11:35:06 [D] context:80           Load memcache for `phycontrib.template.gui.get
_amplitudes`.
11:35:07 [D] context:80           Load memcache for `phycontrib.template.gui.get
_cluster_templates`.
11:35:07 [D] context:80           Load memcache for `phycontrib.template.gui.get
_cluster_pair_features`.
11:35:07 [D] context:80           Load memcache for `phycontrib.template.gui._cl
uster_template_similarities`.
11:35:07 [D] context:80           Load memcache for `phycontrib.template.gui._si
m_ij`.

Aborted!

nsteinme, 2018-05-02:
Could you try deleting the cache folder (F:\SGL_DATA\11 0817\.phy) and try to run it again?

Sarruedi, 2018-05-02:
Thanks Nick! Deleting the cache folder did the trick. The file opens correctly now. Happy Spike Sorting!

nsteinme, 2018-05-02:
great :)

closed 2018-05-02 by nsteinme