cortex-lab/KiloSort #122: Initialize templates 'fromData'?

weallen, 2018-04-16:
I noticed recently that there is an option to initialize the templates from data -- is this recommended? What are the tradeoffs?

marius10p, 2018-04-16:
It is recommended, but we think it's a minor improvement.

marius10p, 2018-04-16:
The trade-off is that it adds run time, and it may or may not run out of RAM (have not checked that recently).

weallen, 2018-04-16:
Thanks. Related to this, is there ever an absolute value for spike detection? I noticed spkTh parameter is set to 6*SD by default -- does this set an absolute bound, or will it ultimately depend on the templates?

marius10p, 2018-04-16:
That is for the threshold crossings that are used to initialize the templates. It's only used when you say "fromData". You can lower it probably to around 4, but not much lower... 

The spike times found by Kilosort are based on a threshold in template space: after projecting raw data onto the template, there is a threshold set in ops.Th. This corresponds to how much of the variance of that spike in the raw data can be explained by the template. Templates that are big on a few channels may explain less variance than templates that are small, but occur on many channels.

closed 2018-08-17 by marius10p