cortex-lab/KiloSort #123: Multiple GPUs

brykko, 2018-04-19:
I'm trying to find out whether there is any gain from installing two GPUs in one machine. Has anyone tried this? Or, does anyone with a better understanding of CUDA know whether it's likely to be beneficial?
Thanks,
Rich

marius10p, 2018-05-15:
Kilosort does not know how to use multiple GPUs. CUDA can probably make integration across GPUs seamless, but the improvement from that is likely to be minimal at this point. Now that GPU prices are back to normal, grab a GTX1080Ti, or wait for 1180. Moving to Matlab R2018a might also see a speed pickup due to better GPU integration.

brykko, 2018-09-10:
Sorry Marius, I only just noticed your response. Thanks for the info!

mnpompili, 2018-11-21:
Hi Marius, 
we are evaluating the same possibility here at the Zugaro lab (using multiple GPU), as we have a dataset from very high density recording and we we run out of memory on the GPU device (1080ti). This also because we  have to increase considerably the ops.Nfilt parameter compared to your guidelines to increase the splitting of clusters and avoid contamination. 
We could batch the spike sorting with subgroups of electrode bundles rather than feeding all of them to kilosort but running every given session together will spare us a lot of time so it would be great to run the algorithm on parallel GPUs  as we have a bunch of 1080ti and 980ti.
Do you think it would be very hard to make Kilosort take advantage of Matlab 'multi-gpu'  ability?
thanks,
Marco

marius10p, 2018-11-21:
Hi Marco,

Sorry, it would be extremely hard and would require a complete re-write. The gains are also likely to be very small, because in addition to the base CUDA code, there is a bunch of Matlab code that cannot be parallelized because Matlab has poor support for state switching between GPUs. 

Wait for Kilosort2, which deals explicitly with drift and you won't need to set Nfilt to be so large. And even if you do set it in the thousands, Kilosort2 has much smaller memory requirements. 

Cheers,
   Marius

mnpompili, 2018-11-22:
Hi Brendon,

thanks for the head up! 
I will give this a try with higher Nfilt settings than what is defined in the config function and see if i get to a point where my clusters are not contaminated