cortex-lab/KiloSort #124: 200 Channel Recordings: Raw Data?

obaiam11, 2018-04-20:
Hello!

I am doing recordings from ~200-300 channels. I only get out raw voltage values, so I write that into a .bin file with size 200 x 300,000 where channel # = 200, and 300,000 represents 15s of data at 20 kHz. I've changed the config file to have these values: 

		
ops.fs                  = 20000;        % sampling rate		(omit if already in chanMap file)
ops.NchanTOT            = 226;           % total number of channels (omit if already in chanMap file)
ops.Nchan               = 226;           % number of active channels (omit if already in chanMap file)
ops.Nfilt               = 452;           % number of clusters to use (2-4 times more than Nchan, should be a multiple of 32)     		
ops.nNeighPC            = 12; % visualization only (Phy): number of channnels to mask the PCs, leave empty to skip (12)		
ops.nNeigh              = 16; % visualization only (Phy): number of neighboring templates to retain projections of (16)		
		
% options for channel whitening		
ops.whitening           = 'full'; % type of whitening (default 'full', for 'noSpikes' set options for spike detection below)		
ops.nSkipCov            = 1; % compute whitening matrix from every N-th batch (1)		
ops.whiteningRange      = 226; % how many channels to whiten together (Inf for whole probe whitening, should be fine if Nchan<=32)		

Is there anything else I should be changing to make this work on my data? When I run the master_file_example file, I receive the following error:

master_file_example_MOVEME
Time   1s. Loading raw data... 
Time   2s. Channel-whitening filters computed. 
Time   2s. Loading raw data and applying filters... 
Time 2.90. Whitened data written to disk... 
Time 2.90. Preprocessing complete!
Index in position 2 exceeds array bounds (must not exceed 452).

Error in fitTemplates (line 42)
            U = cat(3, U, Uinit(:, ipck));

Error in master_file_example_MOVEME (line 20)
rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively

When I look at the DATA variable, it is of size 32832x226x19 in16, and uproj is empty. Inside DATA is all zeros. Any idea how I can adapt my data to work?

Thank you so much!

brykko, 2018-04-20:
I'm guessing here, but it looks likely that the problem is this: `ops.Nfilt = 452`. This value has to be a multiple of 32 otherwise you'll get errors. You could try setting it to 448 or 480, and see if that works?

obaiam11, 2018-04-20:
Hi Richard,

That was my first guess too. I tried that and no luck. Any other things you
might suggest? I am also happy to upload my data to see if you might have
any luck getting it work.

marius10p, 2018-05-15:
Are you writing the binary file with int16 values? 

Might help to post the error after changing ops.Nfilt to a multiple of 32 (does the error change?).

closed 2018-08-17 by marius10p