cortex-lab/KiloSort #125: Setting Channels Inactive

bkarpowicz3, 2018-04-23:
I have data sets with 96 channels of data, and I would like to run individual channels through KiloSort alone.  I am getting the following error: 

```
Index exceeds matrix dimensions.
Error in optimizePeaks (line 119)
    W(:,j,:) = w(:, 1:Nrank)/sum(Sv(1:ops.Nrank).^2).^.5;
Error in fitTemplates (line 32)
        WUinit = optimizePeaks(ops,uproj);%does a scaled kmeans
Error in matToKiloSort (line 31)
rez = fitTemplates(rez, DATA, uproj); % fit templates iteratively 
```

I am attempting to use the following config file.  

`%config  Build the structure of options (ops) for a KiloSort simulation.

ops.GPU         = 1; % whether to run this code on an Nvidia GPU
ops.parfor      = 1; % whether to use parfor to accelerate parts of algorithm
ops.verbose     = 1; % whether to print command line progress
ops.showfigures = 1; % whether to plot figures during optimization

%% options for data storage
ops.datatype = 'dat'; % binary ('dat', 'bin') or 'openEphys'
ops.fbinary  = fbinary; % will be created for 'openEphys'
ops.fproc    = fullfile(fpath, 'temp_wh.dat'); % residual of RAM of preprocessed data
ops.root     = fpath; % 'openEphys' only: where raw files are

%% options for channels and clustering
ops.fs       = 24414;        % sampling rate
ops.NchanTOT = 96; % total number of channels
ops.Nchan    = 4; % num active channels
ops.Nfilt    = 32;  % num clusters to use (2-4x Nchan, multiple of 32)
ops.nNeighPC = 4;           % (Phy): num channels to mask the PCs
ops.nNeigh   = 16;           % (Phy): num neighbor templates to retain projections of

% options for channel whitening
ops.whitening = 'full'; % type of whitening
ops.nSkipCov = 1; % compute whitening matrix from every N-th batch
ops.whiteningRange = 1; % how many channels to whiten together

% define the channel map as a filename made using createChannelMapFile.m
ops.chanMap = [];

% fraction of "noise" templates allowed to span all channel groups
% (see createChannelMapFile for more info).
ops.criterionNoiseChannels = 1;

%% options for controlling model and optimization
ops.Nrank       = 3;     % matrix rank of spike template model
ops.nfullpasses = 6;     % num complete passes through data during optimization
ops.maxFR       = 50000; % maximum number of spikes to extract per batch
ops.fslow       = 3e3;
ops.fshigh      = 300;   % frequency for high pass filtering
ops.ntbuff      = 64;    % samples of symmetrical buffer for whitening and spike detection
ops.scaleproc   = 200;   % int16 scaling of whitened data
ops.NT          = 32*2048+ ops.ntbuff; % batch size (try decreasing if out of memory)
                                       % for GPU should be multiple of 32 + ntbuff

%% options to improve/deteriorate results
% when multiple values are provided for an option, the first two are beginning
% and ending anneal values, the third is the value used in the final pass.
ops.Th  = [4 10 10]; % threshold for detecting spikes on template-filtered data
ops.lam = [5 20 20]; % large means amplitudes are forced around the mean
ops.nannealpasses    = 4;           % should be less than nfullpasses
ops.momentum         = 1./[20 400]; % start with high momentum and anneal
ops.shuffle_clusters = 1;           % allow merges and splits during optimization
ops.mergeT           = .1;          % upper threshold for merging
ops.splitT           = .1;          % lower threshold for splitting
		
%% options for initializing spikes from data
ops.initialize      = 'fromData'; % 'fromData' or 'no'
ops.spkTh           = -1;       % spike threshold in standard deviations
ops.loc_range       = [3  1];     % ranges to detect peaks; plus/minus in time and channel
ops.long_range      = [30  6];    % ranges to detect isolated peaks
ops.maskMaxChannels = 4;          % how many channels to mask up/down
ops.crit            = 0.65;        % upper criterion for discarding spike repeates
ops.nFiltMax        = 10000;      % maximum "unique" spikes to consider

% load predefined principal components (visualization only (Phy): used for features)
dd       = load('PCspikes2.mat'); % you might want to recompute this from your own data
ops.wPCA = dd.Wi(:,1:7);          % PCs

%% options for posthoc merges (under construction)
ops.fracse = 0.1; % binning step along discriminant axis for posthoc merges (in units of sd)
ops.epu = Inf;
ops.ForceMaxRAMforDat = 2e12; % max RAM the algorithm will try to use; will autodetect on Windows`

How can I adjust this file to avoid this error?

marius10p, 2018-05-15:
Please take a look at the file [https://github.com/cortex-lab/KiloSort/blob/master/configFiles/createChannelMapFile.m](createChannelMapFile.m). This shows you how to create a channel map file for your own probe. If you want the channels to be run independently, you need to set kcoords to be a different integer for each channel. The results will still be produced as one single file that you can load into Phy to visualize.

marius10p, 2018-05-15:
Please take a look at the [file](https://github.com/cortex-lab/KiloSort/blob/master/configFiles/createChannelMapFile.m) This shows you how to create a channel map file for your own probe. If you want the channels to be run independently, you need to set kcoords to be a different integer for each channel. The results will still be produced as one single file that you can load into Phy to visualize.

veronikasamborska1994, 2018-06-06:
Hi Marius, 
We have some channels that are dead and were wondering if we could set them as inactive to check if that could potentially help spike sorting? Would setting koords to different integers on these channels do the job? 

Thanks a lot in advance!

nsteinme, 2018-06-06:
Include a new vector in your channel map called "connected" (if it does not
already have this) of the same length as chanMap, xcoords, and ycoords. Set
connected=true for channels you want to include and false for those you
don't.

veronikasamborska1994, 2018-06-06:
Thanks a lot! I am now getting the following error though. I tried changing the number of active channels to 29 (which are the number of those channels marked as true) but still getting the same error.

Error using  + 
Matrix dimensions must agree.

Error in preprocessData (line 159)
            CC        = CC + (datr' * datr)/NT;

Error in master_file_example_MOVEME (line 19)
[rez, DATA, uproj] = preprocessData(ops); % preprocess data and extract spikes for initialization

veronikasamborska1994, 2018-06-06:
Thanks a lot! It worked now and the channels that I marked as dead don't appear in the GUI anymore.  But the waveforms that I am now getting look very weird and all of them look very similar 
[image]
(attached). Do you have any idea why this might be happening?

nsteinme, 2018-06-06:
I think you have set NChanTOT incorrectly. Try downloading neuroscope
(here: http://neurosuite.sourceforge.net/) and try different numbers of
channels until you get it to open properly and show you traces that look
correct. Then use that number for NChanTOT.

veronikasamborska1994, 2018-06-06:
So does this mean I cannot use openephys format when I want to exclude the dead channels? Seems like Kilosort function only imports active channels in the .dat file and I can open it nicely in Neuroscope if I specify the number of channels as that excluding the dead channels.

nsteinme, 2018-06-06:
If you used the provided utility to convert your open ephys file, then the
value that you used for ops.NChan there should be the value that you
subsequently use for ops.NChanTOT, even if you exclude some channels using
connected.

NChanTOT is the number of channels in the datafile itself, regardless of
which ones you will use. When kilosort loads the data, it subsequently
applied the channel map and the "connected" settings, and typically ends up
with fewer channels. ops.NChan (after conversion from openephys) should be
this smaller value representing the number of channels to be analyzed.

veronikasamborska1994, 2018-06-06:
If I understand correctly having 32 channels I should specify both the ops.NChan and ops.NChanTOT as 32 and only change the connected vector in my channel map to indicate which ones are dead.   When I try to do that I get this error 
Error using gpuArray/subsref
Index exceeds matrix dimension.

Error in preprocessData (line 142)
    dataRAW = dataRAW(:, chanMapConn);

Error in master_file_example_MOVEME (line 19)
[rez, DATA, uproj] = preprocessData(ops); % preprocess data and extract spikes for initialization

nsteinme, 2018-06-06:
I think you should try putting a breakpoint on that line with the error
(142) and see what the size of things are and why it has a problem. And/or,
you could share your channel map file. Otherwise I don't really have enough
information to diagnose, sorry -

nsteinme, 2018-06-07:
> Thanks a lot for your help!
> Here is the link to the repository with the masterfile, configfile and
> channelmap that I am using
> https://github.com/veronikasamborska1994/ChannelMap and actually the
> error I am getting is
> After I put a breakpoint it looks like CC matrix is based on the Nchan of
> 32 (CC = zeros( Nchan, Nchan, 'single'), although the datr are based on the
> number of connected channels dataRAW = dataRAW(:, chanMapConn). I am
> guessing this might be causing the problem?

Yes, you have ops.NChan=32, but you should update ops.NChan to be
sum(connected).

veronikasamborska1994, 2018-06-07:
What seems to have worked is loading a .dat file and setting variables this way: 
ops.NchanTOT            = 32;         
ops.Nchan               = 19;          
ops.nNeighPC            = 19; 	
ops.nNeigh              = 19;
The output looks very sensible too and I found the same units as before. 
I have not been able to run it with an OpenEphys file and to update ops.nChan yet 
Thanks for your help though!

veronikasamborska1994, 2018-06-07:
Actually all works now with OpenEphys too. Adding  ops.Nchan = sum(connected) on line 24 of preprocessData.m helped :) 
Thanks again!

lauritk, 2018-06-13:
I had also some trouble with the provided openephys converter, when some of the channels were inactive. The script loaded only first Nchannels files. Had to change the Nchan to NchanTOT in the converter. Otherwise the converter would load wrong channels and leave out some of the correct ones out. Bad channels caused those weird looking waveforms. Script had also some other minor bugs. I can send my version for testing, if you're interested.