cortex-lab/KiloSort #126: Getting stuck in debugging of KiloSort

jvschweim, 2018-04-30:
Hello,
I have been trying to get Kilosort working with my files.

I record in flat binaries, and the script seems to read the file alright. However, I always get stuck at the following point in the preprocessData script.

_Error using  + 
Matrix dimensions must agree.

Error in preprocessData (line 155)
            CC        = CC + (datr' * datr)/NT;_

One of my channels is broken, I should say, but the script finds that. 
I wonder if you have encountered this before, and can help me with that?

Thanks,
Judith

nsteinme, 2018-04-30:
ops.Nfilt must be a multiple of 32 - I think that's probably your issue.
@marius10p, perhaps we can make it automatically set ops.Nfilt to be the
next multiple of 32 from whatever the user has selected.

jvschweim, 2018-05-01:
Hi Nick,
Thanks for your reply, I've just checked the ops.Nfilt is 64, I have attached the workspace data up to the error here.

Maybe it has something to do with the actual datr array just being 33088x31?

I have also attached the ops file, just in case.

[test.zip](https://github.com/cortex-lab/KiloSort/files/1963251/test.zip)

[ops_test.zip](https://github.com/cortex-lab/KiloSort/files/1963258/ops_test.zip)

nsteinme, 2018-05-01:
Ok, looks like your chanmap specifies 31 channels connected, but you have set ops.Nchan to 32. Ops.NChanTOT is the number of rows in the datafile (probably 32, but you should check this too) and ops.NChan is the number of channels that kilosort will be working with (31, in your case). 

@marius10p we can also sanitize inputs for these things. I'm happy to add these bits. Let's discuss tomorrow.

jvschweim, 2018-05-01:
Thanks, I will try this out. I record with 32 channels, so yes I have 32 rows, but one has a bad connections is just noisy. So I assume kilosort recognises this and ignores that channel, is that right?

It is working now, thank you. I will sort through my data and make sure to change the NChan accordingly!

Thank you very much!

closed 2018-05-01 by nsteinme