cortex-lab/KiloSort #128: Running Kilosort

LibiF, 2018-05-02:
Hey,
I installed Kilosort and trying to run it on my data but I keep getting errors for the proprocessData(ops) function.
I think it might be because of the parameters I put in the master file/config_file or how I created a channel map.

Anyone familiar with such problems? OR can send me his config and master files to see ( I am using the template but maybe not changing it correctly)

I also wanted to know which code is run to generate prm files.

Thanks a lot

nsteinme, 2018-05-02:
Hi Libi, I just added example config and master files that will work with
neuropixels phase3A here: https://github.com/cortex-lab/neuropixels/

You will have to set some paths but the parameters should be working, for
that channel map that is also in the repository there. This assumes you
recorded with spikeglx, otherwise it will be a little different. Let me
know how it goes -

LibiF, 2018-05-07:
Thanks,
its working now!

closed 2018-08-17 by marius10p