cortex-lab/KiloSort #132: SVD Error

AimeeJon, 2018-05-16:
Hey,

I'm getting an error with Kilosort during the template optimisation (I think!) - the output is below:

Time 4072.69, batch 781/37266, mu 12.88, neg-err 41234.283484, NTOT 682391, n100 2416, n200 175, n300 27, n400 27
Error using svd
SVD did not converge.
Error in get_svds (line 3)
[Wall, Sv, Uall] = svd(gather_try(dWU), 0);
Error in decompose_dWU (line 17)
        [W(:,:,k), U(:,:,k), mu(k)] = get_svds(dWU(:,:,k), Nrank);
Error in fitTemplates (line 141)
        [W, U, mu, UtU, nu] = decompose_dWU(ops, dWU, Nrank, rez.ops.kcoords);
Error in Kilosorter_ab (line 42)
rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively
Error in batchScriptKS (line 18)
    Kilosorter_ab(fPath,filebase);

(Kilosorter_ab is my own function).
I've run a lot of similar data sets recently and haven't seen this error.  One thing that's different with this set is I've lowered the Nfilt to ~2.3x nChs, rather than the 3x I had for other sets.  I have only had one successful run with the new Nfilt value - could this perhaps be the issue? The data has 5 shanks, with a total of 98 active channels.

Any help you could provide would be very appreciated (:

marius10p, 2018-05-16:
Hi, what is your new Nfilt? This might be related to Nfilt not being a multiple of 32, which it has to be right now. We are working to remove that restriction.

AimeeJon, 2018-05-16:
Hi,
I have a new Nfilt of 224 (down from a previous value of 288), so I think it should be ok?  I was finding the output to be pretty overclustered - hence the decrease.

marius10p, 2018-08-17:
Sorry for dropping this, did you end up figuring it out?

AimeeJon, 2018-08-20:
Hi,
No, I never figured out the root of the issue - I think I just increased Nfilt back to 3xNchs and it worked ok again. Meant things were a little overclustered, but is fairly easy to deal with (:

closed 2018-09-10 by marius10p