cortex-lab/KiloSort #133: Display error when opening file

Sarruedi, 2018-05-17:
I run into a display error with a resorted data set that previously was opening properly using phy after sorting the data with kilosort. I essentially reran kilosort with the data set tweaking some threshold parameters but the output of kilosort seems not to properly open in phy. 
The waveform view, similarity view, pca all do not load and the following error is displayed for phy:

(phy) F:\SGL_DATA>phy template-gui params.py
14:26:11 [E] cli:39               An error has occurred (IndexError): index 2470
20 is out of bounds for axis 1 with size 245980

Does anybody have an idea how this error is caused and how to avoid it?

Best
Sarah

rossant, 2018-05-18:
Maybe try to delete the .phy subdirectory in F:\SGL_DATA?

Sarruedi, 2018-05-18:
Tanks, deleting the .phy did not do the trick but I did resort the data a third time and this time the data loads properly. I guess there must have been an issue at the sorting stage for the previous run.

closed 2018-09-10 by marius10p