cortex-lab/KiloSort #136: Usage of parameters

Alkohost, 2018-05-23:
Hello.

I use Kilosort for high-density 32-channel recordings by silicone probes. I get more or less beautiful clusters with acceptable waveforms, little bit noisy autocorrelograms, but I confused by relatively high firing rate. It could mean I have at least two neurons in the cluster.  I suppose changing some parameters could be helpfull, but have no ideas what to change for better result.

Can you please give more detailed tutorial for params selection?
Thank you.

ops.Th               = [4 10 10];    % threshold for detecting spikes on template-filtered data ([6 12 12])		
ops.lam              = [5 20 20];   % large means amplitudes are forced around the mean ([10 30 30])		
ops.nannealpasses    = 4;            % should be less than nfullpasses (4)		
ops.momentum         = 1./[20 400];  % start with high momentum and anneal (1./[20 1000])		
ops.shuffle_clusters = 1;            % allow merges and splits during optimization (1)		
ops.mergeT           = .1;           % upper threshold for merging (.1)		
ops.splitT           = .1;           % lower threshold for splitting (.1)		
		
% options for initializing spikes from data		
ops.initialize      = 'no'; %'fromData' or 'no'		
ops.spkTh           = -5;      % spike threshold in standard deviations (4)		
ops.loc_range       = [3  1];  % ranges to detect peaks; plus/minus in time and channel ([3 1])		
ops.long_range      = [30  6]; % ranges to detect isolated peaks ([30 6])		
ops.maskMaxChannels = 5;       % how many channels to mask up/down ([5])		
ops.crit            = .65;     % upper criterion for discarding spike repeates (0.65)		
ops.nFiltMax        = 10000;   % maximum "unique" spikes to consider (10000)

nsteinme, 2018-05-24:
If you think Kilosort should have found more clusters, try increasing
ops.Nfilt. You will have to do more merging but it may solve the problem.

marius10p, 2018-08-17:
If the amplitude distributions look like they extend into the noise, then you could benefit from higher threshold ops.Th.

closed 2018-08-17 by marius10p