cortex-lab/KiloSort #137: installation issue

wehrlab, 2018-06-05:
Hi all.
Novice here. 
I'm trying to install KiloSort on Windows 10. When I run 'mex -setup: _path_ C++ in Matlab I get a Warning to update my code :  

"The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. You will be required
	 to update your code to utilize the new API.
	 You can find more information about this at:" 

So when I run mexGPUall I get another Warning: 

"Warning: No CUDA toolkit installation could be found. If installed, set MW_NVCC_PATH environment
variable to location of nvcc compiler."

I installed CUDA from the link provided in installation instructions. Are those unrelated issues?

Am I missing a step here? I looked at updating mex files to use 64 bit api page and it's a little way over my head. How do I find a correct path to nvcc compiler?

thanks,
Ira

marius10p, 2018-06-05:
Hi Ira,

What is your C++ compiler? It should be visual studio (community) 2012 or 2013. 

Alternatively, the CUDA version might be wrong for this version of Matlab, although Matlab would then usually tell you what version to use. My usual advice is to start over with the visual studio and CUDA installation, and restart after installing each. It's a matter of CUDA finding the paths of visual studio, and Matlab finding the paths of CUDA and of visual studio. 

Best,
   Marius

wehrlab, 2018-06-05:
Thanks. Im using Visual Studio 2013 and cuda 9. I'll start over.

marius10p, 2018-06-05:
Try the community version of visual studio....