cortex-lab/KiloSort #138: Error while running

fspinar, 2018-06-05:
Hi all. 
 At first, thank you very much for such an elegant program. I downloaded the Kilosort and can run the master_e_mouse file. However when I try to sort on my own  I get some errors. In the first place, I am not able to use my own data since my data (a mat file derived from a PL2 using Plexon SDK) doesn't have a .wi field, so it is not compatible with the 60-61st sentences of the Standard_Config_MOVEME. When I try to use PCspikes2.mat as data I have:
 _

> _Undefined function or variable 'pm'
> Error in fitTemplates (line 122)
>     Params = double([NT Nfilt Th maxFR 10 Nchan Nrank pm epu nt0]);
> 
> Error in master_file_example_MOVEME (line 20)
> rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively_

_

 I can see that this is because in the fitTemplates function, Nbatch is equal to zero and so the loop between lines 113-119 doesn't work. Yet I couldn't figure out why. Could there be something wrong about my channelMap or directories of my folders? Moreover, why can't I use my own data?
  Thank you and best regards.

nsteinme, 2018-06-06:
Whoops, miscommunication: PCspikes2 and that line of Standard_Config is not the actual data - leave those lines alone for now. 

If you have a matrix of your data that is nchannels x nsamples, then write it to a file like this:
```
fid = fopen('myfilename.dat', 'w');
fwrite(fid, myDataMatrix, 'int16'); 
fclose(fid);
```
Then use myfilename.dat as the data file, i.e. as ops.fbinary.

fspinar, 2018-06-06:
I followed your suggestion and right now it is working properly. Thank you!