cortex-lab/KiloSort #141: Confirming that units are isolated from noise

stephengv, 2018-06-21:
Hi Kilosort Community!

When doing the manual curation in Phy, what objective criterion can one use to decide whether to keep or discard a neural-looking cluster? Presumably a lot of the good-looking clusters I’m seeing are actually poorly isolated from the background noise, but I’m failing to understand how I can make that judgement since the background noise is taken care by kilosort early on and I never get to see it (if I understood your paper correctly). Is there any objective metric of cluster isolation from background noise that I can use?

brendonw1, 2018-06-22:
What about refractory period in the ACG?

marius10p, 2018-06-22:
Background noise is often not the issue. The bigger errors are contamination from other clusters, over-merging and over-splitting, with the latter often happening due to drift. These issues you can usually diagnose by running the wizard and comparing with other clusters found nearby, especially in the template feature view. 

Nonetheless, if you want to tell if a unit is well isolated from the noise, you want to look at its distribution of amplitudes. If it looks "cut off" neatly on the lower end, that means it was separated from noise at the threshold ops.Th. If the distribution trails off in a more gaussian way on the low end, that means it was well above the noise. 

The distribution of amplitudes can be seen either as a function of time, or in the template feature view, as discussed in issue #143.

stephengv, 2018-06-22:
Wonderful! Thanks for the quick response.