cortex-lab/KiloSort #142: Error installing plugin

stephengv, 2018-06-21:
We have been trying to add the  [amplitude histogram](https://github.com/kwikteam/phy-contrib/blob/master/phycontrib/views/amplitude_histogram.py) view to our installation of phy but keep running into an error when we launch Phy, that seems to originate with the attempt to load "AsynchCaller".

[image]

Following the instructions, we have edited the config file, and created a plugin folder in the .phy folder, into which we have placed a text file with the plugin's code, and named that file "amplitude_histogram.py".

Any guidance you can provide would be greatly appreciated.

marius10p, 2018-06-22:
I think you should ask this in the Phy github page, I don't know the answer.

claudiagjou, 2018-06-22:
Thanks! I will do that.

stephengv, 2018-06-22:
Thanks! We will do that.

marius10p, 2018-08-17:
Closing, since this is a Phy issue.

closed 2018-08-17 by marius10p