cortex-lab/KiloSort #145: About Common Average Referencing

LongYangUCLA, 2018-07-03:
I read the common average referencing part in the biorxiv paper. But I can not find this function in the codes. Can anyone please show me where in the code, this function is performed ? Thanks a lot.

nsteinme, 2018-07-03:
Hi, for now you need to implement this operation outside of kilosort. Here
is an example function to do it:
https://github.com/cortex-lab/spikes/blob/master/preprocessing/applyCARtoDat.m

LongYangUCLA, 2018-07-03:
Thanks for your help.

closed 2018-07-03 by LongYangUCLA

rwarren2163, 2018-11-16:
In the biorxiv paper it says that common average referencing is "best performed after high-pass filtering". My understanding is that kilosort performs high pass filtering in addition to whitening, presumably after the user performs common average referencing. Do you recommend users filter, then common average reference data first, then feed into kilosort with filtering turned off? Thanks!

PhantomSpike, 2018-11-26:
That's a good question actually. I think the data collected in the .ap.bin file should in theory be already high-pass filtered at 300Hz in an analog way during collection which is one of the parameters you set in the SpikeGLX. I first do CAR to this data and then feed to Kilosort which does the digital high pass filtering. Haven't compared what happens if you do high pass, then CAR, then feed to Kilosort without filtering there. I have played with different CAR methods but in my experience all of these things don't make a significant difference unless your data is super noisy and needs substantial cleaning. Maybe try both (CRA -> high pass in Kilosort vs high pass -> CAR -> Kilosort filtering off) on a test data and see if there is any difference in terms of cluster quality and downstream analysis. Would be interested to see the results. :)

rwarren2163, 2018-11-27:
Thanks for the thoughtful reply. I am collecting data broadband with an open ephys system. @cxrodgers ran a quick simulation and finds that CAR->filtering is equivalent to filtering->CAR. However, differences emerge when using median rather than average referencing. It's unclear how/whether these differences affect quality, but for me filtering->common median referencing appears to be working well.

I am probably missing something obvious here, but how can I turn off filtering in kilosort (to prevent kilosort from refiltering already filtered data)? I could comment out the calls to filter() in preprocessData(), but perhaps I'm missing a parameter somewhere else? Thanks again.