cortex-lab/KiloSort #147: Matlab loses access to graphics card. TDR issue?

wehrlab, 2018-07-10:
Hi! 
I'm setting up Kilosort to use on windows 10 computer with Nvidia Geforce ti1080 graphics card and Matlab 2018. 

eMouse test ran smoothly without errors, however when I try to run my data I get this error:

_Error using gpuArray
An unexpected error occurred during CUDA execution. The CUDA error was:
CUDA_ERROR_ILLEGAL_ADDRESS

Error in fitTemplates (line 198)
        dataRAW = gpuArray(dat);_

I also get a note from Windows that Application (Matlab I assume) cannot access graphics card. 
After the error I need to restart Matlab to be able to run gpuDevice, otherwise it prints _CUDA_ERROR_UNKNOWN_ which is not very helpful. 
Clustering breaks around 115 seconds on Optimizing templates step. 
I'm trying to cluster data recorded at 30kHz, 64 channels, around ~15 min of recording. 
Originally I thought it was a timeout error, so I disabled TDR on my computer, but it didnt fix it. I also updated the drivers for the graphics card, that didn't fix it either. From what I read, Nvidia geforce cards cannot be switched to TCC mode, which I also tried. What else could this be?

I'll be happy to hear any other suggestions or things to try. 

Thank you in advance,
Iryna

marius10p, 2018-07-10:
That is a standard, non-informative Matlab error when the GPU breaks. Could be anything. First thing would be to check that the data format has been recognized correctly by taking a look at dataRAW (or dat). Plot the channels to make sure they are in the right order etc. Next thing is to check that your number of clusters is a multiple of 32, which we still require, for now. Another possibility is that it couldn't find anything at all on particular batch, maybe because something was wrong on that exact batch. If you can identify which batch it is and stop it before it reaches there, you could see if there is something different with that batch. 

Let me know the results of trying these things?

wehrlab, 2018-07-10:
Thank you, will do!