cortex-lab/KiloSort #148: intermittent error in fitTemplates

ereming, 2018-07-13:
Is there a chance someone has an idea what might have caused this error:

```
Time 1640.78, batch 18481/74376, mu 12.55, neg-err 14465.079625, NTOT 4394352, n100 2413, n200 2413, n300 2413, n400 2413
{Index exceeds matrix dimensions.

Error in fitTemplates (line 212)
        dbins(xround + id * size(dbins,1)) = dbins(xround + id * size(dbins,1))  + 1;

Error in master_file (line 33)
rez = fitTemplates(rez, DATA, uproj);  % fit templates iteratively
}
```

This happened a few times, and then stopped happening. I'm running on a cluster, might it have been a glitch on the system?

marius10p, 2018-07-13:
Could it be there is a batch with zero spikes? I am not sure we protect against that, since it shouldn't normally happen. You could try to debug and check if id is empty... 

Also, your number of clusters has to be a multiple of 32...

ereming, 2018-07-13:
Thanks for the ideas. The number of clusters is a multiple of 64, but I don't know about an empty batch. Would I just check that the value of `nsp` is greater than 0?

marius10p, 2018-07-13:
Check if id and st are empty for that batch.

ereming, 2018-07-13:
Got it. I'll add a try/catch and update if it happens again.

ereming, 2018-07-18:
Got the error again: I forgot to add a check for `id`, but `size(xround)=659x1`, `size(id)=659x1`, and `size(dbins)=[100,64]`.

marius10p, 2018-07-18:
That rules out my hypothesis... Seems like we need to know if max(id) is larger than size(dbins,2) or if max(xround) is larger than size(dbins,1), although I don't have a good hypothesis for why either of these would be true.

ereming, 2018-07-18:
That looks like the problem - `size(dbins,2) = 64` and `max(id)=116`. There are 7/659 values of `id` greater than 63.

marius10p, 2018-07-18:
what is your Nfilt?

ereming, 2018-07-18:
Nfilt is 64.

marius10p, 2018-07-18:
I don't think it changes but do check (look at the size of dWU as well).

id should not be allowed to be >Nfilt. It is surprising that this happens in the middle of your run and not right away on the first batch. What batch did it happen at this time?

ereming, 2018-07-18:
Added a check for dWU. Here's the last output line before the error:

`Time 1444.95, batch 18481/68490, mu 22.20, neg-err 186488.723495, NTOT 6837524, n100 9, n200 9, n300 9, n400 9`

marius10p, 2018-07-18:
Is the error reproducible if you run it again? how much GPU RAM do you have? try to set the batch size smaller just in case (ops.NT)

ereming, 2018-07-19:
It happens intermittently. The computing cluster I'm using has an assortment of GPUs available, so it could the case that the errors are happening with a specific type of card. I haven't figured out how to retrospectively check which GPU past jobs used, but I think I can request specific cards going forward.

marius10p, 2018-07-19:
Ok, maybe that's it then, low GPU memory could result in this. Also, GPU access conflicts, if another program is trying to use the same GPU for compute.

ereming, 2018-08-08:
It looks like this is a bug associated with a particular card (Tesla K20xm) on a particular node on our cluster. The crash happens whenever KiloSort runs on that card, and apparently never on other cards. Strange that it happens in the middle of running, and not right away.

marius10p, 2018-08-08:
Probably because it's an older card and has only 6GB of RAM.