cortex-lab/KiloSort #150: Out of Memory

LongLabGit, 2018-07-23:
After nearly a year of running kilosort on > 150 GB files (with up to 128 channels), I'm running into the following error on a smaller dataset (99 GB .dat file containing 32 channels): 

Out of memory. Type HELP MEMORY for your options.
Error in fullMPMU (line 168)
            rez.cProjPC(irun + (1:numel(st)), :) = gather_try(coefs(iCoefs)'); 

Each time I run it, the error occurs in about the same place (~2200/12000 batches). I've tried decreasing the batch size by half, but that didn't seem to make a difference. (Crashed on batch 4700/24000)

What is the best way to proceed?

marius10p, 2018-07-23:
Seems like it runs out of RAM for cProjPC. Can you please run it with debug mode on and see how large that array is when it crashes? How much RAM do you have, and how many clusters are you using? 

cProjPC is pre-allocated for 5 million spikes, so I wonder if you hit that limit at iteration 2200 and then it has no place to move it to in memory. We could fix that by pre-allocating, but it sounds like you will still run out of RAM if the crash happens at such an early batch.

LongLabGit, 2018-07-24:
I currently have 32GB of RAM, with room to upgrade to 64GB. I'm asking for 176 clusters and expect ~50 usable clusters with > 1 million spikes in each of those 50 clusters. When it crashes, rez.cProj is 92285618x16.

Do you think pre-allocating more spikes and upgrading to 64 GB of RAM will be sufficient?

marius10p, 2018-07-24:
I think it's suspicious you get so many spikes relatively early in the
processing, there is probably a lot of noise. What is your ops.Th? I would
try increasing it.

64 GB of RAM and pre-allocation could work, but I think it will be
borderline.

We can look into writing the feature files to the disk incrementally, but
we'd need to change the writeNPY function for that @nsteinme.

nsteinme, 2018-07-24:
You can't write an NPY file incrementally unless you know the final size
ahead of time, because the array shape goes in the header (and there's no
way to pre-pend a header later so far as I know and I spent a fair amount
of time looking into it...). But you could always write it as a binary
file, then load and re-write incrementally later.

Another solution here btw is to keep fewer nearby channels (e.g. 8 instead
of 16) if the goal is just to reduce the size of cProj.

marius10p, 2018-07-24:
But then you'd also need to reduce cProjPC the same way....

Nick, sounds relatively easy to write the features into a binary file and
then write the NPY file incrementally from the binary. But would Phy have a
problem loading it afterwards? Does Phy load chunks of the feature files,
or the whole thing at once?

rossant, 2018-07-24:
phy uses memory mapping to load large arrays. The OS takes care of loading the requested chunks from disk into RAM, transparently and on-the-fly.

marius10p, 2018-07-24:
Great, thanks Cyrille! Nick or Cyrile, is it possible to add an incremental
version of writeNPY to the npy-matlab repository? Assumes the dimensions
are known ahead of time.

nsteinme, 2018-07-24:
If you wrote to binary, then at the end incrementally loaded and wrote to a
new NPY, wouldn't the NPY at the end be the same as the one you'd have add
if you wrote it all at once like in the current method? Do you just mean,
if you already have a memory problem here, you'll also have a memory
problem in phy? If that's the question, no - phy does not load the whole
thing at once, I don't think.

For writing an NPY incrementally, you just want to do what's in the
writeNPY function -
https://github.com/kwikteam/npy-matlab/blob/master/writeNPY.m - with
constructing the header (now that you know what the final shape will be)
and writing that first - then load and write your data in chunks.

jvschweim, 2018-12-17:
I have a similar issue, when I try to analyse shorter recordings, which are only a few hundred MB compared to my usual 1-2 GB files. Nothing has changed to previous files, just the size of the actual binary file. I haven't quite figured out the exact cut of yet. I don't get an out of memory error, but an Indexing error, see below. But then also the same FullMPMU error afterwards.

Time   0s. Loading raw data... 
Time   2s. Channel-whitening filters computed. 
Time   2s. Loading raw data and applying filters... 
Time 5.31. Whitened data written to disk... 
Time 5.31. Preprocessing complete!
Time   6s. Optimizing templates ...
Time 12.11, batch 601/618, mu 0.00, neg-err NaN, NTOT 1664, n100 13, n200 13, n300 13, n400 13
Time  13s. Running the final template matching pass...
Time 16.61, batch 101/103,  NTOT 0
**Index in position 2 exceeds array bounds.**

Error in fullMPMU (line 205)
[~, isort]      = sort(st3(:,1), 'ascend');

Error in master_file_example (line 27)
rez                = fullMPMU(rez, DATA);% extract final spike times (overlapping extraction)

jvschweim, 2018-12-17:
just an update it processes 550MB files fine, but not 250MB files, same day recordings from same experiments, just shorter recording time.

wuzhengherbert, 2019-08-13:
jvschweim, did you ever figure out how to solve this problem? I'm running into the same issue..

jvschweim, 2019-08-15:
@wuzhengherbert, I have switched to Kilosort2, and so far I haven't had any issues with small files or very large ones either. The spike sorting in Kilosort2 works better for me as well, so that might be worth a try?