cortex-lab/KiloSort #151: Changing Template Window

papokremiotis, 2018-07-25:
Hello, I know that by default the templates look at 61 sample points, but is there a way to change this number? I am working with Aplysia and their action potentials can last 12ms or more and I would like to be able to capture all of it without down sampling my data. Thank you for your help.

marius10p, 2018-07-25:
The option is ops.nt0, but making it significantly larger might have unintended / untested consequences. What do the spikes look like? If they are slowed down version of mammalian spikes, then you're probably better off binning your data. You won't lose detection accuracy, and you will at most lose ~1ms of spike timing accuracy.

marius10p, 2018-08-17:
Any updates, can I close this issue?