cortex-lab/KiloSort #154: Spike time problem

nicklesica, 2018-08-13:
Hi,

There is something strange in my spike times. My data are recorded at 24414.0625 Hz. But when I plot the histogram of the difference  between successive spike times at this resolution, it has strange oscillations at a frequency that is roughly an order of magnitude lower. 

Here is an example histogram:

[image]

This happens in all of my recordings. And it happens for both all spike times and individual clusters.

I haven't seen any mention of this in this forum, so perhaps I am doing something wrong, but I can't imagine what. 

Please help!

Thanks,

Nick

marius10p, 2018-08-13:
Hi Nick,
Just to confirm, these are all the spike times from all neurons, and you are plotting ISI on the x-axis?  That looks like way too many ISIs of 0. I can imagine a few situations: maybe your threshold is set too low (ops.Th), maybe there is some periodic noise at that frequency, or maybe the channel map is not right and the data is not read in correctly. Do the templates look ok? Did you set the number of filters to a multiple of 32? Are there just a few units that contribute to this effect?

nsteinme, 2018-08-13:
Hmmm, I don't see it in my data, e.g.: 
[image]

brendonw1, 2018-08-13:
Aliasing of some sort??

nicklesica, 2018-08-14:
<Sigh>

OK, it was just something dumb. Matlab hist doesn't work for integer types so I was converting the spike times to single precision to make the histogram. But that wasn't sufficient precision so the times were getting somehow quantized. Converting to double precision solved the problem.

Thanks!

closed 2018-08-14 by nicklesica