cortex-lab/KiloSort #156: Template crossing shanks despite kcoords

clausagerskov, 2018-08-17:
Running newest kilosort (commit 0fbe8eb) with the chanMap.mat, master:file and StandardConfig (all files attached) results in templates crossing shanks.

The electrodes used are two 4x4 planer electrodes in two different locations. The final electrode on the second array is used as ground and so is treated as disconnected.
The kcoord variable = [1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 1 2 2 2 2 2 2 2 2 2 2 2 2 2 2 2 NaN];

The two different electrodes should be treated as to independent shanks and be processed separately
[image]

[configfiles.zip](https://github.com/cortex-lab/KiloSort/files/2296819/configfiles.zip)

marius10p, 2018-08-17:
A fraction of them are supposed to cross shanks to account for noise. Default is 0.2, but you can change it.

clausagerskov, 2018-08-17:
Thank you for the quick response!

closed 2018-08-17 by clausagerskov