cortex-lab/KiloSort #157: Applying to three-dimension devices

umiu60135, 2018-08-17:
Hello!

Since the code can only be used for a 2D device now, I am wondering if it is easy to change the code for applying to 3D devices.

Thank you!

 Best,
Zhongpeng

marius10p, 2018-08-17:
Hmm, the position information is mostly used for visualization in Phy, and the underlying algorithm does not depend on it. It would require some changes nonetheless. I wonder though if you truly have a 3D array channels, or whether it could be broken down into multiple shanks isolated from each other. In that case you could use the kcoords parameter to sort the shanks separately.

umiu60135, 2018-08-18:
Thank you so much! However, our channels are based on a kind of very flexible device whose electrodes distribute kind of randomly in space, so it seems not suitable to break down the channels into multiple shanks.

marius10p, 2018-08-18:
Ok, I think Kilosort is ideal for that sort of thing because it doesn't force templates to be localized over channels like all other pipelines do. Instead it uses a low-rank decomposition to regularize waveforms. It also drops any channels with magnitudes less than 1/10 of the peak.

I had a look through the code, and I think the only place we are using x,y coordinates is in the local whitening step. You can avoid it using whitening mode to 'full', which whitens all channels together. If you have a lot of channels though, you might want to figure out how to run this function with 3D channel positions (it needs to know which channels are close together):

https://github.com/cortex-lab/KiloSort/blob/0fbe8ebbae4ebddf44fdfe60bb6b4d9c1888826f/preprocessData.m#L183

Speaking of which, how many channels do you have? If it's too many you might run into other problems too....

umiu60135, 2018-08-18:
Actually we usually have 32 or 64 channels in our devices now. By the way, do you mean that I can't use Phy  by saying "the position information is mostly used for visualization in Phy", or just can't use some functions about the positions in Phy? Can I still refine my results though I don't provide the position information?

Thank you!

marius10p, 2018-08-18:
Phy requires some position information to show waveforms spatially. You can specify it however you want, including as a single column of channels if you don't have any other information. That is actually assumed as default if you don't provide it.

nsteinme, 2018-08-18:
could construct a distance metric between each pair of channels based on how many templates are observed on both of those two channels (and how similar the templates appear on the two channels), and then use multi-dimensional scaling or t-sne or whatever to place the channels in 2D based on that distance matrix? :)

sin-mike, 2018-09-25:
@marius10p @umiu60135 

I do have true 3d recordings of the surface and of the in-depth silicon electrodes simultaneously. What I did is a workaround with different scales. E.g. I changed the scale for the surface electrode 100x times so that the principal group of the surface sites could be connected to the adjacent silicon electrodes. Then I was able to see spike propagation. It didn't affect any of the processing, though, as Phy/Kilosort doesn't require this info for the clusterization. But it was much better for the visualization part. I must admit, that it would be a lot handier to check/uncheck channels for plotting on the fly. 

BR
Mike

umiu60135, 2018-09-25:
Hi Mike,

Thanks for sharing your idea. I'm really interested in it, but I'm sorry I don't totally get it. Could you provide some further physical context? Thank you so much!

Best,
Zhongpeng