cortex-lab/KiloSort #160: Missing spikes

angelonc, 2018-08-29:
There have been several issues bringing this up, but I figured I'll add another one, as it doesn't seem to have been figured out just yet.

It looks like KiloSort is somewhat inconsistently detecting spikes in my data set. Previous issues and comments suggest that this largely applies to particularly large amplitude spikes, but I have been having it happen in my recordings for both large and smaller spikes. See screen shot below from Phy:

[image]

As you can see, these spikes are somewhat small, but spikes just as small or even smaller are detected just fine nearby...

I've attempted to solve this by decreasing the spike detection thresholds from the default of [4 10 10] to [2 10 10], which results in many more spikes being detected, but still misses many of the spikes on this particular channel.

I've also tried increasing Nfilt to the max recommended (64 filts for 16 channels), but still have this issue, qualitatively.

Any suggestions?

angelonc, 2018-08-29:
Here is the same issue, plotted for a 256 channel probe I'm using. In this case, Nfilt is currently 2x the number of channels, to save some time during manual correction, but it is missing both high amplitude and low amplitude spikes...

[image]

marius10p, 2018-09-10:
Hi, 

Kilosort will indeed miss some of the spikes it is unable to cluster. This might be due to a combination of local minima and poor initialization, Have you tried the "fromData" initialization? That initializes templates from k-means on threshold crossings and is generally better at finding more of the neurons. The thresholds you are changing will indeed ensure that in the beginning more waveforms are found, but these might get dropped just as well during the optimization if you keep the final threshold at 10. So maybe try something like [4 8 8]. 

These settings might improve on your issue, but I don't know if it will fully solve it. Kilosort2 will get released in a few months and that solves this problem explicitly by detecting spikes in the residual and introducing them as new templates during the optimization.

angelonc, 2018-09-10:
Hi Marius,

I've tried the fromData option, but found it still is missing spikes (and, anecdotally, using either initialization, it seems like the missing spikes are largely on a particular channel from this recording). I will try your recommendation for the template thresholds.

Thanks again for developing such awesome tools, and can't wait for the release of KiloSort2!

brendonw1, 2018-09-10:
> FWIW: I found that using the following changes to the default settings
> helped a bit for my setup, which is a 16 channel microdrive recording from
> 4 tetrode bundles of nickel-chromium wires gold plated to around 300kOhms:
>
> ops.Nfilt = nchan*4; % (64 channels)
> ops.Th = [4 8 8];
> ops.initialize = 'fromData';
> ops.spkTh = -4;

Thank you Chris for posting this and for sharing what you found to
slightly  help you.  I look forward to KS2's effort to specifically address
this as well.

MarReus, 2019-02-11:
Hi guys!
I'm facing the same problem!

[image]

[image]

I checked and It's not a visualization problem (you can also see it in the screen shots) and I also tried changing all those parameters:
I used a 16 channels silicon prove, and my recordings are around 100 sec long, in awake, behaving animals, so sometimes are a little noisy.
In the begging I used 32 clusters, but then I changed to 64 to try if it was better, but I still have a lot of spikes out of the clustering. 
I also use **ops.initialize = 'fromData'**;  **ops.spkTh = -4**,  and played with **ops.Th = [4 8 8]**; and **ops.lam = [10 30 30]** but still missing a lot. 

I also realized that the remaining recording after subtracting the template (the non colored one) is no as flat as it should be, so I'm assuming something is wrong with the clustering.

Does somebody have any other advise for me? I would really appreciate!
Thanks in advance!