cortex-lab/KiloSort #165: Isolating channels

esiabri, 2018-10-17:
Hi KiloSort team,

Thanks for the fantastic tool.

Is there any way to group the channels so that the algorithm ignores the spatial information and analyze each channel individually? I tried to set the kcoords to [1:Nchannels] but it seems that it still incorporates the data from neighboring channels especially in determining spike times; this sometimes leads to shifting clusters of a single neuron (attached snapshot).
[image]

marius10p, 2018-10-17:
I am not sure at all what happened in your example. However, by default, 20% of the clusters are allowed to exist on all channels so they can capture global noise. 

https://github.com/cortex-lab/KiloSort/blob/b75c1b225b20086a0a736ab03929438c76a93b5f/configFiles/createChannelMapFile.m#L39

You can either set ops.criterionNoiseChannels  to 0 or increase the number of filters, if it's not already 2-4x the number of channels/neurons.

esiabri, 2018-10-17:
still having the similar issue after setting ops.criterionNoiseChannels to 0; this time multiple shifted clusters on channel 9 which seem to be related to the spike times at channel 10.
[image]

For filter numbers more than 64 (I tried 96 and 128), the following error happens opening Phy!
>phy template-gui params.py
11:52:16 [E] cli:39               An error has occurred (AssertionError):

marius10p, 2018-11-17:
Does this happen for all clusters or just a small fraction? How long of a segment of time are you spike sorting? I wonder if the number of channels is not determined correctly. You could visualize that by stopping Kilosort during execution somewhere after preprocessing and looking at the DATA variable.

esiabri, 2018-11-29:
It happens in a small fraction of clusters, but it is consistent across multiple runs with a fixed set of parameters. I have seen these on 10-mins chunks of data on 32 channels.