cortex-lab/KiloSort #167: Spike Time Duplication & Unsuccessful Template Extraction

adehad, 2018-10-25:
Hello All!
### **Overview of the Problem**
When running KiloSort on our single channel data (padded with 3 dummy channels) we observe an unreasonable number of spiking events when the number of templates is below 32. It appears as though the spike_time variable is duplicating each spike event 50 times. Additionally, it seems as though that the unique spike waveforms extracted rapidly drop in similarity to the template after the first few spikes.
This issue may be related to #90, however, that issue does not state that there was an duplication of the spike_times, hence why I created this new issue and it does not explicitly state what steps are needed to fix the issue - hence I could not try to implement it as a solution.
### **Machines / Version Tested On**
We tested on (2) Windows 10 and (1) Linux based machine. Using MATLAB 2018a, CUDA 9.0.
The version of KiloSort tested was both 22-05-18 (0fbe8ebbae4ebddf44fdfe60bb6b4d9c1888826f) and on 10-10-18 (b75c1b225b20086a0a736ab03929438c76a93b5f) (DD-MM-YY format dates). The results were the same.
### **Steps that lead to the Problem**
Our single channel data was converted to 4 channel data (dummy channels of 1s) #22 .
But we felt that the large number of templates causes sub-optimal clustering. We did play around with most parameters, including: `ops.lam, ops.Th`, but the results were not as clean as we would have liked. 
We then tried changing the number of templates, but this caused an explosion in the number of spikes detected.
Old Parameters:
```
ops.Nfilt               = 32;               		
ops.nNeighPC            = 2; 	
ops.nNeigh              = 16;
```

Tried with these new parameters
```
ops.Nfilt               = 4;
ops.nNeighPC            = 2; 	
ops.nNeigh`             = 3;   % As has to be less than Nfilt
```
Additionally had to change: `ops.showfigures = 0;`, as there is a hard coded 9 that prevents using Nfilt less than 9.

### **What we tried so far**
What we initially saw on Phy was incredible. As each waveform looked to match the template very well.
But we saw spikes events in the order of millions, when we extracted the `spiketimes.npy`, things got strange.
Each spike_time had been repeated 50 times. This lead us to believe that the WaveFormView could only display the first few waveforms of the data. This was verified in MATLAB - showing only the first 1000 elements.
[image]
However, things got even more strange, the first few unique 100 waveforms match the template well, thereafter the waveforms start looking more like noise.

Plotting the first 200 unique spike waveforms detected:
[image]

Plotting from the 200th spike to the end of the detected waveforms [i.e. 1315 unique waveforms]:
[image]

The mean of the waveforms highlights how the spikes after the first 200 look like noise:
[image]

The first 200 waveforms match what we saw in one of the clusters, from the `Nfilt` of 32 settings. 
The number of waveforms also looks more reasonable, but still less that what was detected by the `Nfilt` of 32 settings.

We have also used Klusta to get a second opinion of our data:
[image]

Which gives a more reasonable figure for the order of magnitude of spikes detected.

Thank you for your time.

marius10p, 2019-01-28:
Hi, sorry for not catching your message. As you can see in the standard config,

https://github.com/cortex-lab/KiloSort/blob/b75c1b225b20086a0a736ab03929438c76a93b5f/configFiles/StandardConfig_MOVEME.m#L14

 the number of templates has to be a multiple of 32, otherwise unpredictable behavior can and will occur. This is due to the fact that the CUDA code runs in batches of 32 threads. I have removed that requirement in Kilosort2, due to be released soon, but it might be too much work to do it in the old Kilosort. - Marius

adehad, 2019-05-25:
For anyone else's reference, I added the following lines in `rezToPhy` to fix this for my usage:
```
% clean up rez duplicates
[~,uniqueIdxs,~] = unique(rez.st3,'rows');
rez.st3 = rez.st3(uniqueIdxs,:);
rez.cProj = rez.cProj(uniqueIdxs,:);
rez.cProjPC = rez.cProjPC(uniqueIdxs,:,:);
```

This was added before:
https://github.com/cortex-lab/KiloSort/blob/3ff2d1027166b6e54ab4661b8c56e33e1ba9c4e7/finalPass/rezToPhy.m#L29

closed 2019-05-25 by adehad