cortex-lab/KiloSort #168: List Matlab toolboxes needed for Kilosort in Readme

oliche, 2018-10-30:
Hello,
Could there be a round up of the dependencies needed to run Kilosort in the readme ?
So far I can see  Statistics and Machine Learning Toolbox and Parallel Computing Toolbox, but there could be more.

Best Regards

marius10p, 2018-11-17:
Which functions are from the Statistics and Machine Learning Toolbox? I think we are also using signal processing for filtering the data.

oliche, 2018-11-21:
>> make_eMouseData(fpath, useGPU);
To use 'geornd', the following product must be licensed, installed, and enabled:
  Statistics and Machine Learning Toolbox

Error in make_eMouseData (line 44)
    dspks = int64(geornd(1/(fs/fr(j)), ceil(2*fr(j)*t_record),1));

marius10p, 2018-11-21:
I see. You can replace those random numbers with any other positive distribution on the integers as long as you keep mean firing rates roughly the same.