cortex-lab/KiloSort #169: Creating PC Values from data

DJ-Hayden, 2018-10-30:
Hello,

Is there a quick guide for how to create the principal components/ops.wPCA from your own data? In the config files these lines are shown:

```
% options for initializing spikes from data		
ops.initialize      = 'fromData';    %'fromData' or 'no'		

% load predefined principal components (visualization only (Phy): used for features)		
% dd                  = load('PCspikes2.mat'); % you might want to recompute this from your own data		
% ops.wPCA            = dd.Wi(:,1:7);   % PCs 		
```
but I can't find any documentation that explains how to create your own dd.Wi (nor why it only slices the first 7 components).

Thanks!,
Dustin

marius10p, 2018-11-17:
The output should have the same shape as the data in PCspikes2.mat. Extract a bunch of spikes for example by threshold crossings, take 61 time samples, centered at sample 20, then take PCA of those spikes. Only the top 3 PCs are used for Phy visualizations. The only time the PCs are used for actual spike sorting is if you set initialize to "fromData", in which case Kilosort starts with some threshold crossings, projects them into that 7-dimensional PCA space and runs scaled k-means in that subspace.

AurelieP23, 2018-12-11:
Hi, 
Marius, what do you mean by "centered" ?  What feature of the spike(s) should occur at sample 20? 
Is it the first sample of the spike that is more negative than ops.spkTh *sd?
Does it relate to the -21 that is hard-coded in get_PCproj.m ?
Thanks!

marius10p, 2018-12-11:
It's the minimum value of the spike. Yes, it relates to that hard-coded value.

AurelieP23, 2018-12-11:
Ok, got it. Thanks!