cortex-lab/KiloSort #174: eMouse - fwrite error

juplab, 2018-12-05:
Hi, I have used kilosrot successfully on Windows and mac in the past and I am now switching to Ubuntu 16.04. The test eMouse seems to work fine however I have some error when kilosort is trying to save the python results for phy see below:

Error using fwrite
Invalid file identifier. Use fopen to generate a valid file identifier.

Error in writeNPY (line 19)
fwrite(fid, header, 'uint8');

Error in rezToPhy (line 78)
    writeNPY(spikeTimes, fullfile(savePath, 'spike_times.npy'));

Error in master_eMouse (line 37)
rezToPhy(rez, [fpath,'preAutoMerge\']);

It seems that it is a directory set up issue which I am not able to resolve any input would be appreciated
My eMouse directory is: '/home/tom/Git/KiloSort/eMouse'

The modified path in the master_eMouse.m are:

fpath    = '/home/tom/Git/KiloSort/eMouse/'; % where on disk do you want the simulation? ideally and SSD...
if ~exist(fpath, 'dir'); mkdir(fpath); end

% This part adds paths
addpath(genpath('/home/tom/Git/KiloSort')) % path to kilosort folder
addpath(genpath('/home/tom/Git/npy-matlab/npy-matlab')) % path to npy-matlab scripts
pathToYourConfigFile = '/home/tom/Git/KiloSort/eMouse'; % for this example it's ok to leave this path inside the repo, but for your own config file you *must* put it somewhere else!  

Thanks
Tom

brendonw1, 2018-12-05:
Permissions issue for that folder?  Maybe the user you are currently
opening matlab with does not have write permission in the emouse folder.

juplab, 2018-12-05:
eMouse script has been updated and currently instruct the output to subfolders thus the syntax in linux should be / instead of \ in eMouse see below for postAutoMerge (also true for PreAutoMerge)

rezToPhy(rez, [fpath, 'postAutoMerge\']);
in linux this should be rezToPhy(rez, [fpath, 'postAutoMerge/']);

closed 2018-12-05 by juplab