cortex-lab/KiloSort #179: How to use cuda 10 in KiloSort

ShanglinZhou, 2019-02-08:
Hi,

I have a RTX2070 card on my computer. It is only supported by Cuda10. Is there a way I can easily modify the code to use KiloSort with Cuda10? And if yes,  that would be very helpful.

Thank you very much.

marius10p, 2019-02-08:
I haven't tried it yet, but maybe someone else here has. Based on my past GPU upgrades, you will need a matching version of Matlab, probably for the CUDA version. However, you can also install older versions of CUDA with probably minimal loss in performance. Just have to make sure the graphics driver is good, but the driver and CUDA toolkit don't have to be in sync.

ShanglinZhou, 2019-02-08:
Hi Marius,
Thanks for your reply. I guess, since the RTX Card is new turing type crad, only cuda 10 can support it, and lower version of cuda might not work. Then I just installed Cuda10.

I did have a warning "Warning: Version 9.0 of the CUDA toolkit could not be found. If installed, set MW_NVCC_PATH environment variable to location of nvcc compiler.", when I run 'mexGPUall.m'.

But I also get an error "Error using mex No supported compiler was found." It seems my C++ compiler doesn't work well. But if I check the mex -setup C ++, there is a Visual Studio 2013 C++ compiler that is configured. I am a little bit confused by whether it is the problem of cuda version or C++ compiler. Following is the error:

>>mex -setup C++
MEX configured to use 'Microsoft Visual C++ 2013' for C++ language compilation.
Warning: The MATLAB C and Fortran API has changed to support MATLAB
	 variables with more than 2^32-1 elements. You will be required
	 to update your code to utilize the new API.
	 You can find more information about this at:
	 https://www.mathworks.com/help/matlab/matlab_external/upgrading-mex-files-to-use-64-bit-api.html.

>> mexGPUall
Warning: Version 9.0 of the CUDA toolkit could not be found. If installed, set MW_NVCC_PATH environment variable to location of
nvcc compiler. 
> In mexcuda (line 157)
  In mexGPUall (line 4) 

Error using mex
No supported compiler was found. For options, visit https://www.mathworks.com/support/compilers.

Error in mexcuda (line 157)
    [varargout{1:nargout}] = mex(mexArguments{:});

Error in mexGPUall (line 4)
    mexcuda -largeArrayDims mexMPmuFEAT.cu

marius10p, 2019-02-08:
Hi,

Sorry, the Matlab error messages are not  very useful. Matlab has not in fact found CUDA, as that initial warning indicated, and then the compiler did find visual studio, but could not link it with CUDA. 

Downgrade to CUDA 9 please. The code should run fine, see here for similar advice:

https://www.pugetsystems.com/labs/hpc/NVIDIA-RTX-2080-Ti-vs-2080-vs-1080-Ti-vs-Titan-V-TensorFlow-Performance-with-CUDA-10-0-1247/#do-i-need-to-have-cuda-10-to-use-tensorflow-on-nvidia-rtx-20xx-series-gpus

The only alternative is to wait until the March Matlab 2019a release, which will surely come with CUDA 10, but when this kind of tranisition happened before, performance differences were not major between using a current CUDA version and an old CUDA version (i.e. when we all upgraded from CUDA 8 to CUDA9 with the GTX 10xx cards).

ShanglinZhou, 2019-02-09:
Hi,
Thank you very much. I install cuda9, and it works even it says that cuda9 might not  work for my card when I was installing it. 
Thanks for your help again!!!

closed 2019-02-11 by marius10p