cortex-lab/KiloSort #180: Still missing spikes

MarReus, 2019-02-11:
Hi guys!
I'm facing the same problem!

[image]

[image]

I checked and It's not a visualization problem (you can also see it in the screen shots) and I also tried changing all those parameters:
I used a 16 channels silicon prove, and my recordings are around 100 sec long, in awake, behaving animals, so sometimes are a little noisy.
In the begging I used 32 clusters, but then I changed to 64 to try if it was better, but I still have a lot of spikes out of the clustering. 
I also use **ops.initialize = 'fromData'**;  **ops.spkTh = -4**,  and played with **ops.Th = [4 8 8]**; and **ops.lam = [10 30 30]** but still missing a lot. 

I also realized that the remaining recording after subtracting the template (the non colored one) is no as flat as it should be, so I'm assuming something is wrong with the clustering.

Does somebody have any other advise for me? I would really appreciate!
Thanks in advance!

marius10p, 2019-02-11:
Sorry, I don't think we ever figured out where this behavior was coming from. Before the end of the month I will release Kilosort2, and if this problem persists, I can take a look at your data.

angelonc, 2019-02-15:
@marius10p great to hear that KS2 will be released soon! Looking forward to it.

@MarReus Have you tried putting ops.Th even lower? Say [2 6 6]? Not sure if it would help though, because it's missing spikes that have the same amplitude as marked ones.

Dakshitha-BA, 2019-03-18:
@MarReus Did you figure out the problem? I am facing the same issue.

susuchen66, 2019-07-07:
Hi we are facing the same issue with KS2
[image]
[image]