cortex-lab/KiloSort #188: CCgs with significant peak at ~0.1ms

shihaisun, 2019-04-08:
Hi all,

Every now and again I get these odd looking cross-correlograms and I was wondering what you guys think it might be. Below is an example showing this. The CCg has a significant peak at around 0.1-0.2 ms and I don't know how to interpret this. I don't think they are bursting spikes or a monosynpatic interaction between two cells because it is way too short. Could it be a result of template subtraction from KiloSort? or the red cluster is the recording from the axon of blue cluster? Any help would be appreciated.

[image]
Example of odd looking CCg. Window = 10 ms. Bins = 0.1 ms

Thanks,
Scott

marius10p, 2019-04-30:
It is likely due to template subtraction. If it was the axon it would always go off together with soma.

brendonw1, 2019-05-01:
I assume it means it's basically most of the time the same unit.

susuchen66, 2019-07-07:
As Marius has pointed out it's most likely due to new template fit into the residual after template subtraction. so they are literally the same spikes counted twice (fit into different templates). Here is one of our examples.
[image]
[image]