cortex-lab/KiloSort #189: Combine multiple bin files for sorting

isabellow, 2019-04-10:
Hi,

I'm running long behavior sessions and acquiring 3 recordings on a single probe insertion. I'd like to sort all 3 recordings at once, as I expect them to have the same cells (given that I don't move the probe between recordings). Is it possible to input multiple bin files for sorting?

I've tried concatenating the sessions into one bin file, but the resulting matrix is too large for MATLAB to handle.

Any advice would be much appreciated! Thanks!

marius10p, 2019-04-10:
Hi, we don't have that yet, but you can read it sequentially in Matlab and write to a new binary file sequentially. Here is a minimal example:

```
fid_write = fopen(my_binary_file_path, 'w');
for j = 1:length(flist)
    fid_read = fopen(flist{j});
    A = fread(fid_read, '*int16');
    fwrite(fid_write, A, 'int16')
    fclose(fid_read)
end
fclose(fid_write)
```
Btw, consider moving to Kilosort2, since we won't really support Kilosort1 going forward, and the second version should work significantly better for long recordings where small amounts of drift can acumulate.

isabellow, 2019-04-11:
That worked well--thank you! I'll be switching to Kilosort2 for future recordings :)

closed 2019-04-11 by isabellow