cortex-lab/KiloSort #195: Isolating Spikes

aproca, 2019-06-14:
How I would go about isolating individual sorted spikes of the real data (including its waveform, unit, and channel), rather than the templates or the average spikes. Is this in a specific .npy file(s)? 

Thanks!

MarineChap, 2020-01-09:
Dear, 
I am looking for the same thing here. @aproca Did you found an easy way to do it ? :)
Thanks

PhantomSpike, 2020-01-09:
Hi,

You can try this https://djoshea.github.io/neuropixel-utils/

You need to access the raw .bin file and get them from there. The link above should have enough info and worked examples.

MarineChap, 2020-01-09:
Aaah yes, I remember now seeing this during my research... But I don't have access to Matlab. So, I am only working in python with files given as output from kilosort. 
Thanks you. I will try to find a solution through these examples.