cortex-lab/KiloSort #209: rand and randn error in running master_eMouse.m

IljaPilja, 2019-11-26:
Hi,
I'm new at using KiloSort and after having spend too much time trying to install all the required programs, I wanted to verify the installation by trying out master_eMouse.m. The error occurs at line 20 in 'make_eMouseData.m', specifically the function 'rng.m'. Within this function the line 18, namely rand('twister',seed.rand); gives the following error message:

Error using rand
State must be a double scalar or the output of RAND('twister').

The Matlab help seems to suggest the problem arises due to earlier versions of Matlab using such a syntax to control the number generator but should be replaced by alternatives only involving rng. Am I correct in assuming that this is the cause of the error message or does the error arise due to a different problem?
Any advice would be greatly appreciated.

marius10p, 2019-11-26:
What is your Matlab version and full error message please?

IljaPilja, 2019-11-26:
Hi,
this is my version of Matlab:

MATLAB Version: 9.5.0.944444 (R2018b)
MATLAB License Number: 40631714
Operating System: Microsoft Windows 10 Enterprise Version 10.0 (Build 16299)
Java Version: Java 1.8.0_152-b16 with Oracle Corporation Java HotSpot(TM) 64-Bit Server VM mixed mode
-----------------------------------------------------------------------------------------------------
MATLAB                                                Version 9.5         (R2018b)
Brick, also called "fn toolbox" is a set of use...    Version 1.1                 
Image Processing Toolbox                                Version 10.3        (R2018b)
Parallel Computing Toolbox                              Version 6.13        (R2018b)
Signal Processing Toolbox                                Version 8.1          (R2018b)
Statistics and Machine Learning Toolbox          Version 11.4        (R2018b)
Wavelet Toolbox                                              Version 5.1          (R2018b)

The full error message is:

Error using rand
State must be a double scalar or the output of RAND('twister').

Error in rng (line 18)
    rand('twister',seed.rand);

Error in make_eMouseData (line 20)
rng('default');

Error in master_eMouse (line 26)
make_eMouseData(fpath, useGPU);

Sorry not to have mentioned this before. Hope this helps. Thanks.

IljaPilja, 2019-11-30:
Hi,
it seems that I had an additional rng.m file that was not part of the Matlab toolbox was called upon which lead to the error message. One I deleted the path to this faulty rng.m file, the master_eMouse.m did run beyond the line causing the error.
Sorry about that, my mistake.

IljaPilja

closed 2019-11-30 by IljaPilja