cortex-lab/KiloSort #210: Invalid MEX-file: Access is denied.

IljaPilja, 2019-11-30:
Hi,
I'm new to KiloSort and thus the first thing I wanted to try is the master_eMouse.m file to check if everything works well. When I run the mentioned file it displays the following:

master_eMouse

fpath=

    'C:\Users\dsuchane\Documents\Configfiles_KiloSort\'

Time  28s. Loading raw data... 
Time  46s. Channel-whitening filters computed. 
Time  47s. Loading raw data and applying filters... 
Time 73.92. Whitened data written to disk... 
Time 73.92. Preprocessing complete!
Time  79s. Optimizing templates ...

And then stops and the following error message is displayed:

Invalid MEX-file 'C:\Users\dsuchane\Documents\GitHub\KiloSort-master\CUDA\mexMPregMU.mexw64': Access is denied.

.

Error in fitTemplates (line 211)
            mexMPregMU(Params,dataRAW,W,data,UtU,mu, lam .* (20./mu).^2, dWU, nu);

Error in master_eMouse (line 30)
rez                = fitTemplates(rez, DATA, uproj);  % fit templates iteratively

Does anyone know why this error occurs and how to fix it?
Thanks a bunch,

IljaPilja

brendonw1, 2019-11-30:
Permissions?

IljaPilja, 2019-12-01:
Hi,
the only solution is to run Matlab as Administrator, then it works.

closed 2019-12-01 by IljaPilja