cortex-lab/KiloSort #216: Phy WaveformView

justindyao, 2020-02-06:
Hi KiloSort team,

I'm very new to kilosort and just started using Phy to check my data. Currently, WaveformView is displaying "high_pass" versions of the waveforms (waveforms, mean_waveforms, and templates). Below are screenshots of examples.

[image]
[image]
[image]

Is there a way to change the display so that it is not plotting the high_pass versions?

Many thanks in advance,
Justin

rossant, 2020-02-06:
Yes press alt r

justindyao, 2020-02-06:
Thanks. That changed the display text but did not change the waveforms themselves.

rossant, 2020-02-06:
even if you press it repeatedly? 
Can you also update phy and phylib to the latest versions
PS: this issue should actually be on the phy repository

justindyao, 2020-02-06:
Yes, the waveforms still have not updated even if I press alt r repeatedly and after I have updated phy and phylib my terminal lists: 
"actions:204 Error when executing action Switch raw data filter."

it is accompanied with this message:
"FutureWarning, /opt/anaconda3/envs/phy2/lib/python3.7/site-packages/dask/array/core.py:1333: FutureWarning: The `numpy.quantile` function is not implemented by Dask array. You may want to use the da.map_blocks function or something similar to silence this warning. Your code may stop working in a future release." Not sure if they are related.

I tested this out on another computer with an older version and things seem to work appropriately.

Thanks for the help.

Justin

rossant, 2020-02-06:
There was something wrong in the phylib repository. Could you update phylib (master branch) and try again? Sorry for the inconvenience.

justindyao, 2020-02-06:
thanks, it works now!