cortex-lab/KiloSort #219: TemplateGuI kilosort waveform view issue

jnamgoon, 2020-02-26:
Hello,
I am pretty new to Kilosort. I noticed that during manual spike sorting, when a single cluster is selected. some channels would appear empty on the waveform view, even when that channel was listed as showing highest activity. However, when the cluster is selected together with another cluster that shows highest activity on a channel of a different shank, the channel that was initially empty starts showing waves. I know that was phrased very badly so here is an example:
This is  a waveformview of a single cluster with highest activity supposedly at channel 51
[image]

This is when the same cluster (right) is selected with another cluster (left) with supposed activity detected on the channel on a different shank of the electrode:
[image]

Thank you

rossant, 2020-02-27:
Have you installed the latest versions of phy and phylib?

jnamgoon, 2020-03-04:
I believe so.

rossant, 2020-03-04:
The screenshots are from a slightly earlier version, you might need to reinstall phy and phylib. This particular bug was fixed

jnamgoon, 2020-03-06:
I see... thank you very much. I will definitely look into that.