cortex-lab/KiloSort #223: Spike times in rez.st3 exceed data length

udaykj, 2020-06-07:
I've just started using this toolbox on linear probe data and am running into the issue where the computed event times in rez.st3(:,1) go beyond the range of the total number of data samples. I double checked and the sampling rate and datatypes are as they should be. rez.ops.tend/sampsToRead also match the number of samples. For example, for the data I provided, the number of samples is 7995970 whereas the spike times go up to 8060882 (@ 30kHz sampling).

Because of this, when I overlay the spike rasters on the raw voltage trace there is almost no correspondence visually between where spikes seem to be occurring and when the algorithm says they did. I'm wondering if the batch chunking creates some offsets that are not corrected for in the final spike time computation.

Any tips on what the root of this problem may be?

[image]

marius10p, 2020-06-07:
This shouldn't really happen, but I don't maintain this version of Kilosort any more. Please try Kilosort2, and if you get the same problem, I'll try to help.

udaykj, 2020-06-07:
Sorry, I should've mentioned, it was in kilosort2, not kilosort.

marius10p, 2020-06-07:
Please open an issue in the Kilosort2 repository then.

paolahydra, 2022-01-14:
Hi @udaykj 
I have just realized that I have the exact same problem that you described above, with KS2 and sampling rate 30KHz.
I haven't found yet an answer in the KS2 forum. I am wondering if you managed to get to the bottom of this in the end. Thank you for your reply.
Paola

paolahydra, 2022-01-14:
Never mind me... I found my own bug in the concatenation of the .mat file (which is not there for the .dat used by KS, hence the length difference).