cortex-lab/KiloSort #225: Error running Kilosort!

parviz-gh, 2020-08-27:
hi,
i have 2 sessions recording with Neuropixel . i did spike sorting using Kilosort2 separately , all was good. but when i concatenated them using SpikeGLX then run Kilosort2 on concatenated stream the following error was appeared :
'Error running Kilosort! an unexpected error occurred trying to lunch a kernel. CUDA error was:
invalid configuration argument'

JannaAarse, 2021-03-17:
Hey, 

I have the same problem, but not only with concatenated files also with datasets that are recorded continuously (some of them are about 3 hours long) and when I try to run kilosort2.5 with them it takes forever (yesterday the preproccesing step took over 5 hours) and it always crashes when it is about to shift or optimizing the batches. The message I get is the following: 

Error using gpuArray/nan
An unexpected error occurred trying to launch a kernel. The CUDA error was:
invalid configuration argument

Error in median(line 71)
y=nan(s,'like',x)

Error in learnTemplates (line 248)
toc,ibatch,niter,Nfilt,sum(nsp),median(mu),numel(st0),ndrop)

Error in learnAndSolve8b (line 35)
rez= learnTemplates(rez,rez.iorig);

Error in main_kilosort25 (line 46)
rez= learnAndSolve8b(rez,iseed);

How can I fix this problem ? Do you have any idea why this is happening ? 

Best, 
Janna

mahdiramadan, 2022-05-04:
Hi all! Any update on this issue and how to fix it?

JannaAarse, 2022-05-05:
Hey,