cortex-lab/KiloSort #228: first time Kilosort running

Olivier-Dub, 2020-11-02:
Hello,

I am new in Kilosort and I am very exciting to use it (I was previously using SpikeDetect and klustviewa). I recently installed it in Ubuntu 20.04. I wanted to try it on my data (BZ16 in hippocampus) but I have some problems when I try to run main_kilosort. 
My first error is when I want to run datashift2 function:
Index in position 1 exceeds array bounds (must not exceed 7).
Error in getClosestChannels2 (line 15)
iC= isort(1:NchanClosest, :);

Why NchanClosest is fixed at 10? Also, if I fix it to 7 I have more pb after with interp1
Error using matlab.internal.math.interp1
The sample points must be finite.
Error in interp1 (line 188)
 VqLite = matlab.internal.math.interp1(X,V,method,method,Xqcol);

Maybe I did not prepare my ChanMap or my data correctly but they seems OK. Would you have an idea of what is happening?

Thank you very much for your time

Olivier

marius10p, 2020-11-02:
how many channels are you recording from? There are probably a few places where we assume a large number of channels.

Olivier-Dub, 2020-11-02:
I recorded 16 channels. Yes, maybe that is the reason for my error message.

O.

marius10p, 2020-11-02:
How are the channels arranged? Data registration might work for fewer channels too, but it will probably need substantially different settings. I'll probably have to make some updates for that kind of situation.

Olivier-Dub, 2020-11-02:
There are two shanks with 8 channels on each (20-25 micro distance). I tried also with 16 linear probe and I have the same problem.

marius10p, 2020-11-02:
I just realized this is the wrong repo, which is now outdated. It sounds from your message that you are in fact using this one: https://github.com/MouseLand/Kilosort

Olivier-Dub, 2020-11-02:
yes I installed this one. Which version do I need to use?

marius10p, 2020-11-02:
I am saying this github issue is not in the correct repo.