cortex-lab/KiloSort #235: biased against narrow wave-shape spikes?

esiabri, 2021-05-05:
Hi, 
thank you for this amazing tool!
I've noticed that it seems that the narrow wave-shape spikes (coming from putative FS neurons) are more probable to be missed. I have attached here two sample snapshots from the Kilosort GUI, as it can be seen spikes with lower amplitudes were deleted while larger narrower spikes are missed. I was wondering if I miss something and I can fix this through some parameters? does it have something to do with the fact they are less probable to be detected on multiple channels?
[image]
[image]

esiabri, 2021-05-05:
this was with kilosort 3 so I'll move it to the kilosort 3 repository!

closed 2021-05-05 by esiabri

cgallimore25, 2024-08-16:
> this was with kilosort 3 so I'll move it to the kilosort 3 repository!

I'm finding this same issue in my recordings using KS3.  i know this is an old post, but did you ever figure out a combination of parameters that enable KS to detect these spikes?