cortex-lab/KiloSort #237: GPU array error: invalid device function

goenselab, 2021-05-20:
Hi,

Sorry that this is in relation to Kilosort2 but we are using probes with >30um spacing so needs must. I am getting the following errors:

Error using gpuArray/subsasgn
An unexpected error occurred trying to launch a kernel. The CUDA error was:
invalid device function

Error in triageTemplates2 (line 13)
W(:,idrop,:) = []; % remove those templates everywhere

Error in learnTemplates (line 189)
                triageTemplates2(ops, iW, C2C, W, U, dWU, mu, nsp, ndrop);

Error in learnAndSolve8b (line 15)
    rez     = learnTemplates(rez, iorder0);

The GPU is older and it is my suspicion to the source of the error, but any input from anyone else in the know would be much appreciated. GPU on MATLAB 2020b below:

 CUDADevice with properties:

                      Name: 'Quadro K600'
                     Index: 1
         ComputeCapability: '3.0'
            SupportsDouble: 1
             DriverVersion: 11.2000
            ToolkitVersion: 10.2000
        MaxThreadsPerBlock: 1024
          MaxShmemPerBlock: 49152
        MaxThreadBlockSize: [1024 1024 64]
               MaxGridSize: [2.1475e+09 65535 65535]
                 SIMDWidth: 32
               TotalMemory: 1.0737e+09
           AvailableMemory: 796470476
       MultiprocessorCount: 1
              ClockRateKHz: 0
               ComputeMode: 'Default'
      GPUOverlapsTransfers: 1
    KernelExecutionTimeout: 0
          CanMapHostMemory: 1
           DeviceSupported: 1
            DeviceSelected: 1

Thank you in advance!

rodrigomanuelpaz, 2022-02-25:
Hi, have you been able to solve this problem? I'm having the exact same error with kilosort2 and I have the same GPU device properties.

sv-bio, 2022-11-29:
Try to disable stable mode and run mxGPUall.m with this setting.