cortex-lab/KiloSort #239: Unable to launch Phy / Phy2 in update macOS (Version 11)

justin-d-yao, 2021-06-07:
Hi there,

Has anyone else had issues trying to launch Phy or Phy2 after updating their macOS to Big Sur? I keep getting the error of 'Unable to load OpenGL library'. I'm using python 3.9 and I'm guessing there hasn't been any updates to fix this. Curious if anyone managed any workarounds.

Thanks,
Justin

restrepd, 2021-08-05:
I found this thread in stackoverflow:

https://stackoverflow.com/questions/63475461/unable-to-import-opengl-gl-in-python-on-macos

They say: To fix it now you can edit PyOpenGL file OpenGL/platform/ctypesloader.py changing line

    fullName = util.find_library( name )
to

    fullName = '/System/Library/Frameworks/OpenGL.framework/OpenGL'

However, I find that this change has already been made in ctypesloader.py. 

Alternately, they say to use this code:

try:
    import OpenGL as ogl
    try:
        import OpenGL.GL   # this fails in <=2020 versions of Python on OS X 11.x
    except ImportError:
        print('Drat, patching for Big Sur')
        from ctypes import util
        orig_util_find_library = util.find_library
        def new_util_find_library( name ):
            res = orig_util_find_library( name )
            if res: return res
            return '/System/Library/Frameworks/'+name+'.framework/'+name
        util.find_library = new_util_find_library
except ImportError:
    pass

But I do not know where to enter this code in phy 2.

I hope this helps.

restrepd, 2021-08-05:
Never mind. What I stated above did not work because in my phy environment the OpenGL is not found in that directory. Once I moved to my circus environment it worked by calling:

in OpenGL/platform/ctypesloader.py 

fullName = '/System/Library/Frameworks/OpenGL.framework/OpenGL'

phy2 is working again!