cortex-lab/KiloSort #240: Multi-shanks trouble on Phy

mndiguilmi, 2021-06-15:
Hi

I’m currently use a Neuronexus probe A4x4 (16 ch), 4 shanks with 4 channels each one (https://www.neuronexus.com/files/catalog/2021-Probe-Catalog.pdf; page 34). 

Assuming the following distribution of the Neuronexus channels:
(from the left to the right: shank 1 -> 4; channels distribution from down to top: 6, 2, 3, 1 (shank 1); 5, 8, 4, 7 (shank 2) 9, 12, 10, 13 (shank 3); 15, 11, 16, 14 (shank 4))

I made the following channel map for Kilosort 2

function ops=make_16_A4x4_MD_channelMap(fpath,connectedOriginal)
chanMap = [6 2 3 1 5 8 4 7 9 12 10 13 15 11 16 14];
connected=true(16,1);

for i=1:length(connected)
   connected(i)=connectedOriginal(chanMap(i));
end

xcoords=[0 0 0 0 125 125 125 125 250 250 250 250 375 375 375 375];
ycoords=[0 50 100 150 0 50 100 150 0 50 100 150 0 50 100 150];
kcoords = [1 1 1 1 2 2 2 2 3 3 3 3 4 4 4 4];

fs = 20000; 

ops.fs = fs;  
             
ops.NchanTOT = 16;   
        
ops.Nchan = sum(connected);  
         
save(fullfile(fpath, 'chanMap.mat'), 'chanMap', 'connected', 'xcoords', 'ycoords', 'kcoords', 'fs')

When I open the sorted data on Phy, the ProbeView looks OK however all channels appear on the same shank, sh 0 (see below). I’m not really sure if it’s a real problem or not and if it’s a problem in KS or Phy. On the other hand, I’d like to increase the channels for my recording (32-64 ch) using multiple shanks, but I’d like to fix this trouble before.

Thanks!

Mariano

[image]

kingsEffy, 2024-01-29:
just to wave in seems the kcoords isnt saved in channel map related npy files but these:'
    chanMap0ind = int32([1:rez.ops.Nchan]-1);
    writeNPY(chanMap0ind, fullfile(savePath, 'channel_map.npy'));
    writeNPY([xcoords ycoords], fullfile(savePath, 'channel_positions.npy'));
', so .. shankid is inaccuarte? 

besides I'm having the same issue, also 4 shanks...

Lorena-J, 2024-04-02:
Hi! I have a similar issue, i'm trying to do the sorting with a 4-shank probe, all the coordenates are revised and ok, but when i plot the probe view is not the same shape than my probe. 
[image]
[image]
[Páginas desdeCambridge-NeuroTech-Product-Catalog.pdf](https://github.com/cortex-lab/KiloSort/files/14836597/Paginas.desdeCambridge-NeuroTech-Product-Catalog.pdf)